qmk/qmk_firmware #4404: qmk doesn't boot on STM32F103 bluepill

milestogo, 2018-11-10:
This is a poor bug report because I'm baffled. 

Latest trunk, clean download of QMK on clean install of ubuntu , chibios_test/stm32_f103_onekey builds fine. Flashed to the device, the symptom is that nothing happens and it doesn't seem to initialize USB.  When I put some extra LED blinking in the matrix functions, nothing happens there.  

Doing the same thing with the almost identical TMK stm32_f103_onekey works fine. It blinks & I have a working keyboard. 

I've ruled out hardware by reproducing this on multiple devices. TMK works, QMK doesn't. 
Same behavior on different host OSes or through a USB hub. 

I've tried symlinking qmk's copy of chibios to the TMK one. It builds, same problem. 
I can't see anything wrong in the board configuration and all the makefiles & hal information seems to be identical between 
https://github.com/qmk/qmk_firmware/tree/master/keyboards/chibios_test/stm32_f103_onekey 
and 
https://github.com/tmk/tmk_keyboard/tree/master/keyboard/stm32_f103_onekey

This isn't a secondary bootloader problem, I'm flashing directly to 0x8000000 on the device with an STlink adapter. 
For giggles I've tried using QMK with the arduino bootloader. 

I'd love suggestions on older versions of the code that are likely to work, sections where QMK may be changing chibios defaults,  or of hardware debugging I can provide.

awkannan1, 2018-11-10:
What version of the arm tool chain are you on? If it’s the version provided by aptitude as part of 18.04, there are issues with it. I’m cutting a PR soon to fix the install script for arm on Ubuntu.

milestogo, 2018-11-10:
That sounds promising.  I'm on Ubuntu 18.04.1 

gcc-arm-none-eabi                     15:6.3.1+svn253039-1build1           amd64        GCC cross compiler for ARM Cortex-A/R/M processors
libstdc++-arm-none-eabi-newlib        15:6.3.1+svn253039-1+10 

(and breadcrumbs for the next person who reads this. There's an ARM debugging guide at https://beta.docs.qmk.fm/for-makers-and-modders/arm_debugging)

milestogo, 2018-11-10:
confirmed that it's the Ubuntu toolchain. I just built with the Eclipse-mcu toolchain described in the arm debugging doc & the binary works as expected.

milestogo, 2018-11-14:
Duplicate of # 4209.

closed 2018-11-14 by milestogo