qmk/qmk_firmware #4408: Send key when held on timer

eletoblade, 2018-11-11:
I am attempting to send a certain key (tab, arrow keys) on an interval that I can specify. There does not appear to be a way to do this within the process_record_user framework, instead I had to go to the old action_get_macro functionality and by using the matrix scan. I would prefer not to use the action_get_macro as SAFE_RANGE cannot be used with it since that causes the key code assigned to be greater than the uint8 can handle. Does anyone know of a way to do this in the process_record_user framework? I can think of a couple ways to attack this (below) but haven't been able to figure out how to implement any of them. Screenshots of the code below.

1. something similar to process_record_user that runs when a key is released.
2. some way to test in the matrix scan if the tab key is held.

drashna, 2018-11-13:
well, not sure why you reverted to the action_get_macro.   But you'd have to use matrix_scan_user in either case here.

eletoblade, 2018-11-13:
sorry, I should have been clearer. Right now what I am doing is using action_get_macro to set a variable and timer when the key is pressed. Then in matrix_scan_user I am using if statements to test the state of the variables and then hit the key. 

The reason I am using action_get_macro is because process_record_user has no way (that I cold find) to run a code when a key is released so I am not able to wipe these variables and just get stuck in an infinite loop. What I am looking for is a way to wipe these variables with a process_record_user like function so that I can use SAFE_RANGE in my keycode enum. The matrix_scan_user makes sense to me, it is having to do it the old action_get_macro that doesn't.

Please let me know if this doesn't make sense or if the code would be helpful.

drashna, 2018-11-13:
It does though.  `record->event.pressed`. If it's true, it's pressed, if it's false, it's released.

noroadsleft, 2018-11-14:
@eletoblade Here's a code example of using `process_record_user` and `record->event.pressed` to execute actions on press and release.

The `if` block runs when the key is pressed; the `else` block runs is when the key is released.

https://github.com/qmk/qmk_firmware/blob/e10a602e7b1ac8d7f2ff027ae41570fe1a2ba57d/keyboards/lets_split/keymaps/default/keymap.c#L181-L190

eletoblade, 2018-11-15:
ok, thanks. I just realized the reason that wasn't working for me before was I had a process->event.pressed before the case block. Thanks for the help, I appreciate it.

closed 2018-11-15 by eletoblade

drashna, 2018-11-15:
Ah, yeah, that would do it.  And .... I hate that specific method/setup. 

But glad to help!