qmk/qmk_firmware #4415: Replace default boot loader on daisy

duhanebel, 2018-11-13:
Hi,
I'm not sure how to replace the default boot loader on a daisy v1 board so that I can use qmk. 
The board has an atmega32u4 and I've noticed the file `utils/bootloader_atmega32u4_1_0_0.hex` but there are no instruction on how to load that.

drashna, 2018-11-13:
There is actually instructions on how to use it, but if you don't know the term or how, it's not obvious.

Specifically, you cannot just flash a new bootloader. You need to ISP Flash it to the board.  Not just that, but you may need to change the fuses on the controller, as well. 
https://docs.qmk.fm/#/isp_flashing_guide

Or more relevant: 
https://www.reddit.com/r/olkb/comments/8sxgzb/replace_pro_micro_bootloader_with_qmk_dfu/

duhanebel, 2018-11-13:
Ah, thanks! For some reason I didn't see that page, probably because I was not searching for "ISP".
From what I see in the first guide there are two steps:
1. Buy an ISP flasher or flash the ISP firmware into an Arduino pro micro to hack together one
2. use the flasher at step 1 to update the board's Bootloader
is that correct?

Is there a way I can make sure I'm flashing the right thing? I just looked at the chip on the keyboard and it says: atmega32u4 so that's why I was pretty sure I was on the right track.

drashna, 2018-11-14:
Yup, that's what I mean. :)

And yeah, those are the steps that you'd want to take. 

And yeah, if that's the what the chip says, then you should be good.

closed 2018-12-15 by drashna

gwerbin, 2019-02-10:
Can the info about fuses be added to https://docs.qmk.fm/#/isp_flashing_guide ?