qmk/qmk_firmware #4424: Combo feature can result in transposed output

noahfrederick, 2018-11-14:
The combo feature results in transposition of key presses for me under certain circumstances. Steps to reproduce are below.

As an aside, I think a robust implementation of combos has really great potential, and although there are a number of issues that make it unusable for me currently, I'm excited for its future (to say the least).

-----

## Steps to reproduce

In my `config.h` I set:

```c
#define COMBO_COUNT 1
#define COMBO_TERM 100
```

I doubt the `COMBO_TERM` value actually matters, but I include it here just in case.

Then take the example from the documentation:

```c
const uint16_t PROGMEM test_combo[] = {KC_A, KC_B, COMBO_END};
combo_t key_combos[COMBO_COUNT] = {COMBO(test_combo, KC_ESC)};
```

Now type the word `as` (possibly has to be typed within `COMBO_TERM` ms). Note that `a` belongs to a combo and that `s` does not—I believe this matters.

### Expected output

`as`

### Actual output

`sa`

drashna, 2018-12-09:
Mhh, that seems like it's a timing issue, but I'm not sure how one would fix this.

onetom, 2019-01-24:
This feature made me the most excited about getting a 30% minidox keyboard from falba.tech, which is just absolutely marvelous in every aspect.
However this ordering issue makes using combos impossible and the learning curve is much higher.
Here is my config, which also exhibits issues with W & E or I & O being swapped when typed quickly:
https://github.com/onetom/qmk_firmware/commit/d48e7a84751b9f3c861dc656bb6509d9138cad70

onetom, 2019-01-24:
This ordering phenomena has been observed in Dec 28, 2016 already:
https://github.com/qmk/qmk_firmware/pull/960#issuecomment-269439454

There is a PR which enhances combo behavior and references the comment above, but it DOES NOT solve the ordering issue:
https://github.com/qmk/qmk_firmware/pull/2561

rsheldiii, 2019-01-25:
@onetom are there any extra repro instructions to see out of order keys? I have the code from #2561 running on my gherkin with [similar combos](https://gist.github.com/rsheldiii/8eb5dcf873000153a5327318171991a9) and `we` comes out in the correct order for me

noahfrederick, 2019-04-10:
I can confirm that this particular issue is fixed after https://github.com/qmk/qmk_firmware/pull/2561.

closed 2019-04-10 by noahfrederick