qmk/qmk_firmware #4516: i want to only use half of my 60% nyquist.....

zavjo, 2018-11-29:
so i bought the nyquist PCB but i only need half of it. but im having problems programming it. ? i am using the right half.

drashna, 2018-11-30:
Okay, if you used the QMK Configurator, that makes things difficult.

This makes it assume that "master left", which means it uses the left half of the layout, but will reverse it, because you're using the right half (and it assumes it's the left, so it flips it).

zavjo, 2018-11-30:
so should i not use qmk config. whats your advice on what i should use ? @drashna

drashna, 2018-12-09:
You can use the QMK configurator, it is just a lot more confusing to do so.

In that case, you'd want to only set one half of it, and then .... I think you'd need to mirror that half, so that it will map to the correct positions.

closed 2019-11-21 by drashna