qmk/qmk_firmware #4517: Caps-lock LED woes xd87

WarmCatUK, 2018-11-29:
I obtained an xd87 and intended on running qmk on it. Someone beat me to it but hadn't added caps-lock led support.
Caps led seems to be on pin E2.
Led flashes upon keyboard being plugged in, but led doesn't toggle with caps lock (stays off).
Initially assumed the issue was only under osx, however installed windows 10 on bootcamp and issue persists.
Here's the code in xd87 I've added:
`void led_set_kb(uint8_t usb_led) {
    // put your keyboard LED indicator (ex: Caps Lock LED) toggling code here
    if (usb_led & (1 << USB_LED_CAPS_LOCK)) {
        //register_code(KC_C); //to test if this actually gets executed (it doesnt)
        PORTE &= ~(1 << 2);
    } else {
        //register_code(KC_O); //to test if this actually gets executed (it doesnt)
        PORTE |= (1 << 2);
    }
  led_set_user(usb_led);
}`

Here is the file: https://github.com/WarmCatUK/qmk_firmware/blob/messingabout/keyboards/xd87/xd87.c

WarmCatUK, 2018-11-30:
I can build a layout with qmkeyboard.cn which toggles the led no problem, so it must be something silly I'm missing out or doing wrong.

fougner, 2018-12-02:
I remember having some issues with this when playing around with it. Unfortunately I have it at work so I can't try stuff out, but I can check it during the week.
@WarmCatUK IIRC all background lighting leds were connected to a MOSFET and I don't remember seeing any special handling for the capslock led, how is it connected hardware wise? Have you verified your code actually executes with e.g. a print statement and use hid_listen to check?

WarmCatUK, 2018-12-02:
It’s connected directly to E2. I don’t think the code is executing.
I’ll look into how to using hid_listen and give that a go :-)

fougner, 2018-12-03:
https://github.com/fougner/qmk_firmware/commits/xd87-capslock

this seems to be working for me. That's a pretty old parent though, so something might've changed in upstreams master since then

fougner, 2018-12-03:
So I did a bisect to figure out why it doesnt work on latest master. Seems this commit broke it. https://github.com/qmk/qmk_firmware/commit/39bd760faf2666e91d6dc5b199f02fa3206c6acd

@mechmerlin anything changed in how the led functions are called?

fougner, 2018-12-04:
Let's close this as a dupe of https://github.com/qmk/qmk_firmware/issues/4471 then. 

Seems like people agree on the faulty commit.

WarmCatUK, 2018-12-05:
Yep good idea, can discuss further on #4471

commit d6b7ca04 (2018-12-05): Only try to read the report ID from SetReport when the keyboard is part of the shared EP

Fixes #4471
Fixes #4517
closed 2018-12-05 by WarmCatUK