qmk/qmk_firmware #4535: Error building handwired/frenchdev

matthewbauer, 2018-12-02:
I'm getting this error on handwired/frenchdev:

```
$ make handwired/frenchdev:default
keyboards/handwired/frenchdev/frenchdev.c: In function 'init_mcp23018':
keyboards/handwired/frenchdev/frenchdev.c:57:24: warning: increment of a boolean expression [-Wbool-operation]
         i2c_initialized++;
                        ^~
 |
 | .build/obj_handwired_frenchdev_default/keyboards/handwired/frenchdev/frenchdev.o: In function `frenchdev_blink_all_leds':
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:31: undefined reference to `frenchdev_led_all_off'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:32: undefined reference to `frenchdev_led_all_set'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:33: undefined reference to `frenchdev_led_1_on'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:35: undefined reference to `frenchdev_led_2_on'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:37: undefined reference to `frenchdev_led_3_on'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:39: undefined reference to `frenchdev_led_1_off'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:41: undefined reference to `frenchdev_led_2_off'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:43: undefined reference to `frenchdev_led_3_off'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/frenchdev.c:44: undefined reference to `frenchdev_led_all_off'
 | .build/obj_handwired_frenchdev_default/keyboards/handwired/frenchdev/keymaps/default/keymap.o: In function `matrix_scan_user':
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/keymaps/default/keymap.c:377: undefined reference to `frenchdev_led_1_off'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/keymaps/default/keymap.c:378: undefined reference to `frenchdev_led_2_off'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/keymaps/default/keymap.c:384: undefined reference to `frenchdev_led_1_on'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/keymaps/default/keymap.c:387: undefined reference to `frenchdev_led_1_on'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/keymaps/default/keymap.c:388: undefined reference to `frenchdev_led_2_on'
 | .build/obj_handwired_frenchdev_default/keyboards/handwired/frenchdev/keymaps/default/keymap.o: In function `led_set_user':
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/keymaps/default/keymap.c:402: undefined reference to `frenchdev_led_3_on'
 | /tmp/nix-build-qmk_firmware-0.6.144.drv-1/source/keyboards/handwired/frenchdev/keymaps/default/keymap.c:404: undefined reference to `frenchdev_led_3_off'
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [tmk_core/rules.mk:286: .build/handwired_frenchdev_default.elf] Error 1
Making handwired/gamenum with keymap default                                                           [OK]
```

Can anyone reproduce this?

drashna, 2018-12-09:
Nope.

What version of avr gcc are you using?  If it's 8.2, that may be why.
You can find out by running `avr-gcc --version`

matthewbauer, 2018-12-09:
This is on gcc 7.3.0 and binutils 2.30.

drashna, 2018-12-11:
what version of QMK firmware are you compiling on? 
And have you updated your repo recently?

drashna, 2019-11-21:
Whatever issue was here has since been fixed.

closed 2019-11-21 by drashna