qmk/qmk_firmware #4548: F12 - 24 Keys not working on Massdrop CTRL

M1K8, 2018-12-04:
When assigning any keycode between F13-F24, nothing happens when they key is pressed. This happens with both the online configurator when manually building the firmware. register_code() and unregister_code() are also non-functional with these keycodes.

drashna, 2018-12-09:
I think this may actually be an OS issue, and not keyboard/QMK related.

Are you using something to keep track of the outputs and seeing  if the OS is receiving anything, at all?

M1K8, 2018-12-09:
Yep, and the OS isn't receiving anything. Strangely enough if I use another qmk keyboard, the additional f keys work fine. This issue also persists across multiples machines

M1K8, 2018-12-10:
The keyboards where the f keys work fine are using a AT90USB646 and a 32u4, whilst the problematic keyboard is using a ATSAMD51J18A.

fauxpark, 2018-12-10:
This is probably because the ATSAM HID report descriptor [only declares usages up to `0x65`](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/arm_atsam/usb/udi_hid_kbd.c#L112), and the F13-F24 keys are `0x68` through `0x73`. If you change the Usage Maximum and Logical Maximum to `0xFF` it should work.

M1K8, 2018-12-13:
That fixed it, thanks!

patrickmt, 2018-12-19:
@M1K8 just to be clear, did you increase both `Usage Maximum` and `Logical Maximum` to `0xFF` as suggested by @fauxpark for the correct functionality?

PR #5180 (2019-04-12): Fix ATSAM's USB HID descriptor to support higher heycodes
## Description

Assuming I understand this correctly, this should set the max scancode to E7 (RGUI), which is the last listed code in the main list (everything else use for internal QMK/TMK stuff).

## Types of Changes

- [X] Core
- [ ] Bugfix
- [ ] New feature
- [X] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

*  Fixes #4548, partially addresses #4597 

## Checklist

- [X] My code follows the code style of this  [... cut]
closed 2019-04-12 by skullydazed