qmk/qmk_firmware #4549: Allow overriding rules

Arkonos, 2018-12-04:
I want to set `UNICODEMAP_ENABLE = yes` in my own ergodox layout. I added this line to my rules.mk and compiled. I got the error:
`quantum/quantum_keycodes.h:85:6: error: #error "Cannot enable both UNICODEMAP && UNICODE"`
I checked keyboards/ergodox_ez and line 80 reads:
`UNICODE_ENABLE = yes`
Compiling succeeded after I commented this line out.

I think it would be useful if the more specific rule set overruled its parents.
Should this not be desired or possible, I'd suggest expanding the unicode documentation.

Arkonos, 2018-12-04:
Unicodes are still not working, I'll reopen when they do and I'm sure that this is necessary.

closed 2018-12-04 by Arkonos

drashna, 2018-12-04:
Yeah, this is normal.  You need to add this to your rules.mk:
```make
UNICODE_ENABLE = no
UNICODEMAP_ENABLE = yes
```