qmk/qmk_firmware #4578: Tx Bolt ghosting second character on key press

sdothum, 2018-12-08:
Commit c0baf2a964b10d708281c704c0b049a1cf0f4914 on Sat Nov 18 09:38:15 2017 -0600 causes Tx Bolt mode to ghost a second character on most key presses (confirmed on Planck and Splitography keyboards). The ghosted character is always the same character from the layout matrix per key, though, is from a unique relative position to the key pressed -- that is, there is no obvious discernible pattern to the ghosted keys.

The ghosted keys are such, KEYPRESS:output

    S:s   T:tw  P:pk  H:hs  *:*p  *:*h  F:ft  P:p   L:l   T:td  D:d
    S:sh  K:kp  W:wt  R:r   *:*k  *:*w  R:rs  B:b   G:g   S:s   Z:z 

The last working commit for Tx Bolt is 5f4c2dfd84467dc7f04e8e07c294ebfa5b4ca459T

drashna, 2018-12-09:
Well, the simple solution in the short term would be to revert that commit, and see if it fixes it.

eg `git revert c0baf2a`

xsrvmy, 2018-12-10:
Can confirm that version 0.6.4 does not have this issue. Hopefully this gets fixed soon.

drashna, 2018-12-10:
@xsrvmy Could you try running the git command above on your local branch, and see if that fixes the issue?

xsrvmy, 2018-12-11:
@drashna I reverted to a slightly earlier commit with a version number on it and it's working properly.

drashna, 2018-12-11:
/sigh
`git revert` is not going back to a specific point in time.  It actually reverts the change that the commit made, essentially removing it. 

So the reason I asked about that command specifically, is to see if that change was the cause.  And from there, we can see if we can figure out what went wrong. 

Because otherwise, your using code that is a year out of date, and missing a lot of fixes.

xsrvmy, 2018-12-11:
I checked out master and did the revert and it works fine.

drashna, 2018-12-11:
Awesome.  Then it's some change to that code that is causing the issue then.

germ, 2019-01-26:
Has there been any update on this? Will see if I can repro with a Gergo

tschulte, 2019-02-22:
Maybe the author of that commit, @seebs, should have a look into this. This renders TX Bolt mode useless and we have to use Gemini mode.

seebs, 2019-02-22:
Huh, interesting.

Oh, hey.

```
+static bool update_state_bolt(uint8_t key, bool press) {
+  uint8_t boltcode = pgm_read_byte(boltmap + key);
+  if (press) {
+    state[TXB_GET_GROUP(boltcode)] |= boltcode;
+    chord[TXB_GET_GROUP(boltcode)] |= boltcode;
+  } else {
+    state[TXB_GET_GROUP(boltcode)] &= ~boltcode;
+  }
```
This seems suspicious, but I can't immediately see the failure.

What's interesting to me is the difference between the two S keys. Both of those have the same entry in boltmap, in principle? So we should get boltcode = TXB_S_L, which is 0b1. TXB_GET_GROUP() should be yielding 0, so we should |= 0b1 into both chord[0] and state[0], and on release, we should & those bits out. The tw/pk cases look almost like swapping bits -- it's 0b001000 and 0b000100 being
flipped, and similarly 0b010000 and 0b000010, and s/h  would be 0b100000 and 0b000001. All specifically in the case where the top bits are 0s. Hmm.

When nothing's still pressed, we should send the chord, which means that we should have 0b1 in chord[0], so we'll send that. And... I don't see how this produces an h. h is 0b100000, and nothing should be masking it in. But weirdly, we get it apparently?

I can try to debug this, I have a keyboard that supports this functionality and I have plover. I don't see any obvious way this makes sense, though.

seebs, 2019-02-22:
oh for !@#*!s sake

```
      switch(mode) {
        case STENO_MODE_BOLT:
          update_state_bolt(keycode - QK_STENO, IS_PRESSED(record->event));
        case STENO_MODE_GEMINI:
          update_state_gemini(keycode - QK_STENO, IS_PRESSED(record->event));
      }
```

the reason we're seeing weird glitches that relate to the bits in the raw keypress is that there's no `break;` here and thus it's falling through into the gemini case, and setting one arbitrary extra bit in one of six bytes. my bad.
to fix: add "break;" after `update_state_bolt(...)`.

yanfali, 2019-02-23:
Fall through ftw

tschulte, 2019-02-23:
I will prepare a pull request now.

closed 2019-02-24 by drashna

seebs, 2019-02-24:
thanks much, very sorry about that. that was deeply mystifying. i spent way too much time staring at the fancy bitwise logic without realizing that it was a distraction.

(There's a secondary question about the results in the `state` variable, because those won't have the top two bits of each byte set correctly, but I suspect users checking that state wouldn't care, because they know which byte to check already. I don't know.)

PR #5229 (2019-02-24): Fix Tx Bolt ghosting second character on key press
## Description

Fix missing break; in switch statement.

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #4578, Fixes #4506 

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [x] I have read the [**CONTRIBUTING** documen [... cut]
commit a69e4406 (2019-02-24): Fix Tx Bolt ghosting second character on key press (#5229)

* convert tabs to spaces

* fix #4578: don't call gemini protocol code when in bolt mode

Add missing break; statements in switch. The missing break resulted in
a fall through and an additional processing of the gemini code.