qmk/qmk_firmware #4595: K-Type second USB port no connectivity

manno, 2018-12-10:
I'm trying to connect the K-Type via the second (right) USB-C port, but it doesn't work.
The left port works fine, I can type.

Is this hard to fix? Where would I look?

lbibass, 2019-01-02:
That seems more like an issue of the board, not the firmware.

manno, 2019-01-02:
I can test it again. Works with other firmware, though.

Looking at the [K-Type BOM](https://github.com/kiibohd/pcb/blob/master/K-Type/K-Type%20PCBA%20BOM.xlsx), I guess it's this chip TS3DS10224RUKR?

ishtob, 2019-01-02:
is [this](https://github.com/kiibohd/controller/blob/b9fe845c943868e49fd26b685760b6aba1ae8bfe/Scan/K-Type/scancode_map.kll#L164)potentially what we need to check on initiation and select the usb port?

closed 2019-12-21 by stale[bot]

bradyjoh, 2022-06-22:
I want to +1 this, I just compiled the latest QMK and flashed it to the K-Type using `dfu-util` and the left USB port under the ESC key works, but the right USB port under Pause does not allow the keyboard to show up in Devices, though its RGB does light up. This is with `qmk` version 1.1.0

```
QMK Firmware 0.17.0
Making input_club/k_type with keymap default

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

```

For reference, I can go back and flash my KLL firmware and the keyboard goes back to registering both ports identically.

QMK firmware and KLL firmware attached for reference. [KLL QMK Firmwares.zip](https://github.com/qmk/qmk_firmware/files/8957924/KLL.QMK.Firmwares.zip)

orogenic, 2024-08-19:
I flashed an up-to-date compile of `-kb input_club/k_type -km default` on linux 6.10.5. When I plug in the right USB, I *only* get a "Cannot enable. Maybe the USB cable is bad?" error message every ~4 seconds. When I plug in the left USB, I see the keyboard registered, *and* I get the "Cannot enable. Maybe the USB cable is bad?" error message every 4 seconds.

I don't know how to debug this.