qmk/qmk_firmware #4597: HID report descriptors are not homogeneous

fauxpark, 2018-12-10:
Taking a look at the HID report descriptors on the various platforms supported, I noticed differences between them. For example, the 6KRO descriptor for the ATSAM protocol only declares up to keycode `0x65`, meaning F13-F24, locking caps/num/scroll lock, and international keys all do not work (#4548). On the other hand, LUFA and V-USB both correctly declare up to `0xFF`.

It would be nice to unify these descriptors as much as possible so that all platforms produce the same (or similar) sequence of bytes.

drashna, 2018-12-11:
Where do you find those for LUFA and V-USB? 
Also, what about ChibiOS?

fauxpark, 2018-12-11:
Here are all the descriptors I could find. ChibiOS seems to just run LUFA.

**ATSAM**
[Keyboard](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/arm_atsam/usb/udi_hid_kbd.c#L97)
[NKRO](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/arm_atsam/usb/udi_hid_kbd.c#L259)
[Extra](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/arm_atsam/usb/udi_hid_kbd.c#L427)
[Mouse](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/arm_atsam/usb/udi_hid_kbd.c#L588)
[Raw](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/arm_atsam/usb/udi_hid_kbd.c#L740)
[Console](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/arm_atsam/usb/udi_hid_kbd.c#L886)

**MBED**
[Keyboard](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/mbed/HIDKeyboard.cpp#L66)

**PJRC**
[Keyboard](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/pjrc/usb.c#L166)
[NKRO](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/pjrc/usb.c#L201)
[Mouse](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/pjrc/usb.c#L242)
[Debug](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/pjrc/usb.c#L293)
[Extra](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/pjrc/usb.c#L309)

**USB HID**
[Mouse, Keyboard, Raw](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/usb_hid/arduino-1.0.1/cores/arduino/HID.cpp#L47)

**V-USB** (no NKRO?)
[Keyboard](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/vusb/vusb.c#L264)
[Mouse, Extra](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/vusb/vusb.c#L307)

**LUFA**
[Keyboard](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/usb_descriptor.c#L54)
[Mouse](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/usb_descriptor.c#L102)
[Extra, NKRO](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/usb_descriptor.c#L160)
[Raw](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/usb_descriptor.c#L229)
[Console](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/usb_descriptor.c#L251)

drashna, 2019-02-19:
Since this is stuff that is a bit beyond me, do you have time to take a look at this, and get the settings in sync.  at least, as much as possible? 

If not, would there be any issue with merging this:
https://github.com/qmk/qmk_firmware/compare/master...drashna:fix_hid_report_descriptors
@patrickmt

fauxpark, 2019-02-19:
LGTM :)

I also forgot to mention #4871 is related and partially fixes this issue for LUFA.

patrickmt, 2019-02-19:
Yes this seems good to go!

closed 2019-08-13 by drashna

fauxpark, 2019-08-13:
@drashna this issue is still not resolved, I would very much like to see all the descriptors I listed above be aligned.