qmk/qmk_firmware #4610: Develop some sort of include system for keycode documentation

jetpacktuxedo, 2018-12-12:
## Feature Request Type

- [ ] Core Functionality
- [ ] Add-on hardware support (e.g. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing Feature(s)
- [ :heavy_check_mark: ] New behavior

## Description 
As discussed in discord, it would be really nice to have some sort of system for generating https://docs.qmk.fm/#/keycodes and all of the keycodes on the `Features` pages from a single source. This would help prevent those pages from falling out of sync with one another and make it easier to update with new keycodes.

jakevandervaate, 2022-03-25:
Docsify has an [embed](https://docsify.js.org/#/embed-files) feature that allows embedding contents of files into other files.

**Example:**

Creating a file called "keycodes_modifiers.md", and placing the table with the modifier keycodes in it, would allow for that file to be embedded into the [Full List of keys page](https://docs.qmk.fm/#/keycodes?id=modifiers) and the [Modifier Keys page](https://docs.qmk.fm/#/feature_advanced_keycodes?id=modifier-keys).

Reference "keycodes_modifiers.md" in any other file with

```
[keycodes_modifiers.md](./keycodes_modifiers.md ':include')
```

This would provide a single source for any given set of keycodes, and would prevent different pages from having different lists of keycodes (like the two pages I linked above).

I've tested this on my branch with the files listed below:

* A file with modifier keycodes: [keycodes_modifiers.md](https://github.com/jakevandervaate/qmk_firmware/blob/docs-embed-keycodes/docs/keycodes_modifiers.md)
* A file with keycodes_modifiers.md embedded in it: [keycodes.md (full list of keys)](https://github.com/jakevandervaate/qmk_firmware/blob/docs-embed-keycodes/docs/keycodes.md#modifiers-idmodifiers)
* Another file with keycodes_modifiers.md embedded in it: [feature_advanced_keycodes.md (modifier keys)](https://github.com/jakevandervaate/qmk_firmware/blob/docs-embed-keycodes/docs/feature_advanced_keycodes.md#modifier-keys-idmodifier-keys)

This builds and displays correctly on my system when I run `qmk docs` and view the local version.

I'd be happy to apply this change to the rest of the keycode types, and submit a PR, if this is good solution for this issue.

Erovia, 2022-03-25:
I think this is a step in the right direction. Could you please open the PR? Feel free to ping me on it.

Anika-Roy, 2022-10-11:
Hi! Im participating in Hacktoberfest and I'd like to know if this issue is counted in Hacktoberfest and if I can work on it?

3geek14, 2023-08-15:
Is there a reason the linked PR stalled? Is this issue looking for a new approach?