qmk/qmk_firmware #4611: Mod-Tap with toggle layer

ollyhayes, 2018-12-12:
Not sure if this is a bug or not, maybe it is not even possible, but I'm trying to have my left shift key work as ordinary shift when held down, and toggle layer 1 on when it is pressed.

I've tried setting it as `MT(MOD_LSFT, TG(SYMB))` and as `MOD_LSFT(TG(SYMB))` and neither work. 

 - Keyboard:  ergodox_ez
 - Operating System: Ubuntu
 - avr-gcc version: 5.4
 - arm gcc version: 6.3.1
 - QMK Firmware version: 0.6.179
 - Any keyboard related software installed? None

drashna, 2018-12-12:
You can't do this. Unfortunately. 

MT only support basic keycodes, for the keycode. 
If it's not on this list, it won't work:
https://docs.qmk.fm/#/keycodes_basic

This is because of the size that QMK uses for keycodes.  There is only enough room to support 0xFF, which means that only standard keycodes are supported (QMK functions are above 0xFF).

ollyhayes, 2018-12-12:
Ah okay that makes sense.

What about the other way around? Would it be possible to have a layer switching key that works like `LT(layer, kc)` that toggles to the layer when pressed but activates `KC_LSFT` (or `MOD_LSFT`?) when held down?

drashna, 2018-12-12:
Well, LM sort of does that.

But you could use the `layer_state_set_user` function to accomplish this, as well.

ollyhayes, 2018-12-12:
I think with `LM` the layer you're switching to is only activated whilst you're holding it down though right? I'd like the layer to be on until I switch to anyther layer specifically.

I will look into `layer_state_set_user` though. Thanks for your quick responses.

ollyhayes, 2018-12-13:
Ah that macro looks perfect, I'll give it a go later. Thanks very much.

I'm still not understanding what you're saying about `LM` and `MO` though. You say 'Momentary toggle', but as far as I can see there is no toggle involved, it seems to work like a push and hold button. As soon as you let go you're back to the layer you started on. The [docs](https://docs.qmk.fm/#/feature_advanced_keycodes?id=switching-and-toggling-layers) confirm that:

> `MO(layer)` - momentarily activates layer. As soon as you let go of the key, the layer is deactivated.

ollyhayes, 2018-12-15:
Ah that works perfectly, thanks for your code. In the end I did it like this: (`register_mods(MOD_BIT(...))`)

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
	static uint16_t fnx_layer_timer;

	if (keycode == KC_FNX) {
		if(record->event.pressed){
				fnx_layer_timer = timer_read();
				register_mods(MOD_BIT(KC_LSFT));

			} else {
				unregister_mods(MOD_BIT(KC_LSFT));

				if (timer_elapsed(fnx_layer_timer) < TAPPING_TERM)
					layer_invert(NORMAL);
		}
		return false;
	}
```

closed 2018-12-15 by ollyhayes

ollyhayes, 2018-12-15:
Actually I found that if I was too fast at typing a shift + key combination it would still switch to the normal layer so I changed it to this:

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
static uint16_t fnx_layer_timer;
static bool fnx_inverting;

if (keycode == KC_FNX) {
	if(record->event.pressed){
			fnx_layer_timer = timer_read();
			fnx_inverting = true;
			register_mods(MOD_BIT(KC_LSFT));

		} else {
			unregister_mods(MOD_BIT(KC_LSFT));

			if (timer_elapsed(fnx_layer_timer) < TAPPING_TERM && fnx_inverting)
				layer_invert(NORMAL);
	}
	return false;
}

fnx_inverting = false;
```

danielo515, 2019-07-25:
So the thing here is that, when you held the shift it acts as shift, but when you just hit it you go tho that layer? And then how you come back from it?

ollyhayes, 2019-07-27:
I had another key on that layer that I was switching to that did `TO(BASE)` to take me back to the base layer.

To be honest though I don't use that layout anymore. All my layers use `MO(LAYER)` now.

danielo515, 2019-07-27:
Using MO you need to hold it right? 
I've been reading the docs for invert layer and seems that tapping the same key again will "revert" that layer, switching back to the original.

ollyhayes, 2019-07-27:
Correct, holding down `MO(LAYER)` switches to that layer whilst it's held down, and then goes back to the layer you were previously on.

Not sure what you mean by "invert layer", but it sound like you're talking about toggle layer, `TG(LAYER)`. Pressing it once will switch to that layer, pressing it again takes you back.

danielo515, 2019-07-27:
I mean the part of the code: `layer_invert(NORMAL)`