qmk/qmk_firmware #4616: MUV_IN and MUV_DE are entirely undocumented

jetpacktuxedo, 2018-12-12:
The source for them is [here](https://github.com/qmk/qmk_firmware/blob/ade22f8e2c272044ea2f80ff6fe5ca9576858939/quantum/process_keycode/process_audio.c#L40-L50) so I figured it out, but should still probably be documented on the [keycodes page](https://docs.qmk.fm/#/keycodes)

danielo515, 2018-12-19:
You may start writing on this issue what do you think they are for.

jetpacktuxedo, 2018-12-19:
According to the source (linked in the initial issue comment) they seen to cycle "voices" forward and backwards

joecabezas, 2019-06-09:
I still have no idea what this does

fauxpark, 2019-06-09:
Basically, they change a few aspects of the sound emitted from the speaker (in particular the timbre). You can see what all the voices do here: https://github.com/qmk/qmk_firmware/blob/master/quantum/audio/voices.c

PompTart, 2021-12-21:
At any rate, it doesn't seem to work on my Planck

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc