qmk/qmk_firmware #4625: EEPROM support for Massdrop arm_atsam (CTRL & ALT)

helluvamatt, 2018-12-13:
Per [the datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/60001507C.pdf) for ATSAMD51J18A the MCU has a [SmartEEPROM function](http://ww1.microchip.com/downloads/en/DeviceDoc/60001507C.pdf#_OPENTOPIC_TOC_PROCESSING_d115e202581) that allows up to 64K of EEPROM storage.

I am working on implementing this in `tmk_core/common/arm_atsam/eeprom.c` as that file is already set up to handle eeprom simulation in memory and it should be fairly simple to configure SmartEEPROM.

drashna, 2018-12-14:
Well, check out the STM stuff too.  I think that has some of the same limitations/issues that ATSAM has, so it would be a good place to look.

tprk77, 2019-06-25:
I've been watching this ticket, can I offer to help in some way?

drashna, 2019-08-22:
There is a PR that adds persistent storage.  So it maybe worth checking that out.

daltona, 2019-08-22:
Well I'm sorry I missed this issue, but my implementation is actually implementing SmartEEPROM support, with a fallback to original functionality if SmartEEPROM is not enabled on the chip.
Enabling SmartEEPROM can be done with a change that I've made to mdloader, referenced in #6068 
So for sure I believe it exactly solves this issue.

zvecr, 2021-12-17:
Fixed with 6068.

closed 2021-12-17 by zvecr