qmk/qmk_firmware #4627: Encoder not working with Pro Micro

BenRoe, 2018-12-13:
I try to get the encoder working on a breadboard with a Pro Micro. Added the code from [feature_encoders.md](https://beta.docs.qmk.fm/features/feature_encoders) and use the Pins PD1 (2) & PD4 (4).

[image]

**Describe the bug**

The key is not triggered if i rotate the encoder.

**System Information**
 - Keyboard:  Breadboard
   - Revision (if applicable):  
 - Operating System: Windows 10

Full code 
https://github.com/BenRoe/qmk_firmware/tree/breadboard-test/keyboards/breadboard

nooges, 2018-12-18:
I ran into the same issue with 32u4, if I change the lines in encoder.c involving `readPin()` by casting to bool as follows, the encoder works:

Old: `(readPin(encoders_pad_a[i]) << 0) | (readPin(encoders_pad_b[i]) << 1)`

New: `((bool)readPin(encoders_pad_a[i]) << 0) | ((bool)readPin(encoders_pad_b[i]) << 1)`

nooges, 2018-12-20:
Should be fixed now that https://github.com/qmk/qmk_firmware/pull/4681 has been merged in.

BenRoe, 2018-12-27:
Tried it today, but still not working.

closed 2019-02-14 by BenRoe

LouWii, 2019-06-23:
This bug is still present. I'm using a ProMicro and an encoder on pins D2 and D3.

The detection was really messy. Either the firmware was "detecting" multiple change per second. Or none. Depending on the encoder position.

After using @nooges fix, the encoder behaves properly.

PR #5130 (2019-02-15): Improve Encoder documentation

## Description

## Types of changes

- [ ] Core
- [ ] Bugfix
- [ ] New Feature
- [ ] Enhancement/Optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/Layout/Userspace (addition or update)
- [X] Documentation

## Issues Fixed or Closed by this PR

* Fixes #4627

## Checklist:

- [ ] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [X] I have updated the documentation accordingly.
- [X] I have read the **CONTRIBUTING** document. (https://docs.qmk.fm/#/contributing)
- [ ] I have added tests to cover m [... cut]