qmk/qmk_firmware #4629: OSL not working properly when autoshift is enabled

danielo515, 2018-12-13:
**Describe the bug**

When I have the autoshift enabled (hold long for shift a key) the OSL modifiers are not working as expecting. 
What I expect is to hit te OSL, go to the desired layer, hit one keystroke and come back. However, if the key I hit is not exculded from autoshift then it does not come back to the original layer, I have to hit one of the keys excluded from autoshift (in my case letters) to get out of the layer.

Also, OSL keys started to behave weird, in the sense that I can not hit again to deactivate the layer, i have to hit it several times once it finally decides to come back to the original layer, usually I have to tap it twice in a row and then on the third tap it comes back.
I have the ONESHOT_TAP_TOGGLE set to two.

Regards

**System Information**
 - Keyboard:  ergodox
   - Revision (if applicable):  ez
 - Operating System: windows
 - QMK Firmware version: 0.5.174
 - Any keyboard related software installed? 
  - none

drashna, 2018-12-13:
Yeah, this is probably normal. 

Specifically, the autoshift function returns false, at one point.  So this would block the One Shot layer from clearing properly. 

On this line:
https://github.com/qmk/qmk_firmware/blob/master/quantum/process_keycode/process_auto_shift.c#L197

Try adding this before the `return false;` and see if that fixes the issue:
```c
#ifndef NO_ACTION_ONESHOT
set_oneshot_layer(ONESHOT_PRESSED);
#endif 
```

danielo515, 2018-12-14:
Hello @drashna , thank you for your prompt response.
When I add your code I get this error while compiiling:

```
ycode/process_auto_shift.c: In function 'process_auto_shift':
quantum/process_keycode/process_auto_shift.c:198:9: error: too few arguments to function 'set_oneshot_layer'
         set_oneshot_layer(ONESHOT_PRESSED);
         ^
```

drashna, 2018-12-14:
That's what I get for not testing this

drashna, 2018-12-14:
The command should actually be `clear_oneshot_layer_state(ONESHOT_PRESSED);`

danielo515, 2018-12-14:
Ook, that makes more sense. I'll try it out. I tried looking at the documentation, but I didn't found that particular function, just the cancel one and I was not sure about it

drashna, 2018-12-14:
Yeah, the issue is that the function calls `return false`, so it doesn't process the keypress properly.

I had this issue with my custom macros, too.  The simple solution is to return true, but that DOES NOT work here.  So, you have to tell one_shots, that it's been pressed.

danielo515, 2018-12-15:
Hello @drashna ,
With the suggested change it now compiles, but the behavior remains the same. I don't know if this makes a difference, but I'm using autoshift for numbers and symbols.

This is the code I wrote at the lines you specified

```

        autoshift_on(keycode);
      #if !defined(NO_ACTION_ONESHOT) && !defined(NO_ACTION_TAPPING)
        clear_oneshot_layer_state(ONESHOT_PRESSED);
      #endif
        return false;

      default:
        autoshift_flush();
        return true;
```

apalugniok, 2019-02-14:
I have also just experienced this issue. Doing this worked for me. Not sure if this is how it's meant to be solved.
```
autoshift_on(keycode);
      #if !defined(NO_ACTION_ONESHOT) && !defined(NO_ACTION_TAPPING)
        clear_oneshot_layer_state(ONESHOT_OTHER_KEY_PRESSED);
      #endif
        return false;

      default:
        autoshift_flush();
        return true;
```

drashna, 2019-06-04:
Double checking the code, that is the correct solution.

PR #6067 (2019-11-03): Smallish overhaul of Auto-Shift feature
* Fix edge case when using One Shot Layer with Auto Shift, and it not triggering the cleanup
* Remove junk code (no longer used)
* Replace `(un)register_code` calls with `tap_code` where appropriate
* Fixed up Switch check to be more readable (less verbose)
* Simplified modifier check (if it comes back non-zero, there are mods)
* Add additional function calls for autoshift settings
* Made all variables static, since there are function calls to get their status
* Fixed up documentation

## Types of Changes
- [x] Core
- [x] Bugfix
- [x] Enhancement/optimiz [... cut]
closed 2019-11-03 by drashna