qmk/qmk_firmware #4637: "TAPPING_TERM" is unclear

fauxpark, 2018-12-14:
As a native English speaker, I find the name `TAPPING_TERM` to be kind of unhelpful as to its purpose. Would it make more sense to call it something like `TAP_TIMEOUT` or even `TAP_TIMEOUT_MS`? There is already a precedent for this naming in `LEADER_TIMEOUT` and `ONESHOT_TIMEOUT`.

`TAPPING_TOGGLE` could also be eg. `TOGGLE_TAPS` or `TAPS_TO_TOGGLE`.

drashna, 2018-12-14:
uuuuuuugh. 

I mean, yeah, the name isn't exactly clear.  But considering it's wide usage ..... have you searched for `TAPPING_TERM` in the repo? 

331 results in 221 files.

fauxpark, 2018-12-14:
I did, and I see where you're coming from, but it's only a simple string replace, no meaningful code changes. I guess your concern is that the keyboard directories are more or less the responsibility of their maintainers?

End of the day, it's the most minor of minor things so if it really isn't worth all that...

drashna, 2018-12-14:
That's part of it.

The other part, is finding a meaningful name that isn't prone to the same issue. 

But `TAPPING_TERM` is used anywhere that is looking to determine the timing for a tap vs a hold.  That's ... a lot of different functions.  For instance, Auto Shift, Tap Dancing, Mod Tap, layer tap, and One Shot keys all use `TAPPING_TERM` in some way or another. 

`ONESHOT_TIMEOUT` is different. It's how much time until a one shot key "times out" and turns off.  Eg, setting this to 500 means that you have 500 ms to hit another key, before it automatically reverts. 

`LEADER_TIMEOUT` is also handled differently. It's how long you have TOTAL to hit the leader sequence, which can be  up to 3 keys, IIRC.  So .... a low value for this will make it impossible to use. 

`TAP_TIMEOUT` doesn't really seem to make sense in a lot of this. I think that this would actually be a worse name for it. 
`TAPPING_TOGGLE` is mostly used for the `TT` (Tap Toggle) layer keycode, so changing that isn't a good idea. 

But don't get me wrong, a discussion is a good thing.  And I may just be stuck on using that term, since I've gotten used to it. So it may be hard for me to see another name.

fauxpark, 2018-12-14:
> `TAPPING_TERM` is used anywhere that is looking to determine the timing for a tap vs a hold. That's ... a lot of different functions.

Sure, but... it's literally just a name. Heck, you could temporarily make `TAPPING_TERM` an alias of `TAP_TIMEOUT` and not have to change them all in one go. (admittedly, that is a little messy, and probably one more thing for reviewers to look out for)

> TAP_TIMEOUT doesn't really seem to make sense in a lot of this. I think that this would actually be a worse name for it.

It makes sense to me; as you said above it controls the length of time at which a potential "tap" will become a definite "hold" action. So, if I understand correctly, when I press a key and release it within `TAPPING_TERM` milliseconds, it will perform whatever the tap action is. If I keep it held past `TAPPING_TERM`, it will then immediately perform the hold action. The window you get for performing a tap has timed out.

fauxpark, 2022-09-26:
Closing, not really sure what to do with this.

closed 2022-09-26 by fauxpark