qmk/qmk_firmware #4644: usb serial rgb control?

acme64, 2018-12-14:
I requested if the guys at the aurora-project (https://github.com/antonpup/Aurora) could support QMK firmware, they mentioned you guys would need USB serial RGB control support? does that sounds accurate?

drashna, 2018-12-14:
Well, the ZealPC boards actually support "RAW HID" for this, and can be controlled computer side, in theory.

Though, QMK already has support for a virtual serial port, but not rgb control over it. 

So, I suppose it's entirely possible.

helluvamatt, 2018-12-16:
I've had a look at the Aurora source code. They're a WPF app that communicates with devices through native VC++ "wrapper" DLLs.

One would have to write a wrapper DLL to communicate with QMK through some RAW HID protocol.

Any device with support for the RGB Matrix component could be adapted to use the RAW HID protocol to set LEDs. I've got code for the Massdrop CTRL in my [ctrl_matt](https://github.com/helluvamatt/qmk_firmware/tree/ctrl_matt) branch.

DavidCerrone, 2019-05-05:
I currently have this implemented for my macro keyboard with the accompanying WPF app. It will be open sourced an PR'd into the QMK repo when the development is complete. You can check out the product [here](https://nullbits.co/db1pro/). This includes settings RGB and macro settings via the app.

VCJames, 2019-06-01:
DavidCerrone - I have some STM32 chips out for delivery right now and have been researching reverse engineering popular keyboard RGB protocols like Corsair - cannot thank you enough for your efforts.

wanildopneto, 2020-09-20:
So... Has there been any progress by now?

shawnz, 2020-09-20:
I feel like there is a big need right now for a common, open protocol for controlling RGB lighting effects over USB. It seems like QMK is in the perfect position to provide that and it would be a big selling point for this firmware. Please consider adding this feature to the base distribution

harrismirza, 2020-10-14:
Would MIDI not work for this? I'm new to QMK so I'm not sure if it supports being both a keyboard and a MIDI device at the same time, but if it is possible then you could use Midi input, you can send 2 bytes of data per standard message, which is restrictive, but there is also SysEx messages which AFAIK don't have a set maximum size.

mxmilkiib, 2022-06-18:
It would be nice if this discussion continued

tzarc, 2022-06-18:
#11567.

zvecr, 2022-08-28:
Closing in favour of the above XAP ticket.

closed 2022-08-28 by zvecr