qmk/qmk_firmware #4668: Some DZ60 layouts are obsolete.

Micro-Biology, 2018-12-16:
After discussion [here](https://github.com/qmk/qmk_firmware/pull/4658), it became apparent the DZ60 layouts need consolidating.

There are multiple layouts that could easily use another layout but with some keys doing nothing, eg the HHKB layouts or layouts with keys that are not split.

I am proposing that the layouts are reduced to:

1. ANSI_layout (currently Layout)
2. ISO_layout (as above but with an ISO enter, in the pull request linked above)
3. ANSI_Arrow (currently LAYOUT_60_b_ANSI)
4. ISO_Arrow (Currently LAYOUT_60_b_ISO)
5. ANSI_2U_Lshift (Layout_all)
6. ISO_2U_Lshift (as above but with ISO enter, currently does not exist)

This has the advantage of all possible layouts being supported whilst reducing the number of layout macros.

This would however cause many/all the DZ60 keymaps to be invalid. I am inexperienced in both qmk and github issues and so would like some advice as to how to go about making these changes.

Micro-Biology, 2018-12-16:
First 2 changes have been made.

On info.json the legends are not perfect, however I am under the impression as they are just layouts it doesn't matter too much?

noroadsleft, 2018-12-16:
The legends in the `info.json` file do not matter. They're only for people to use when debugging the file.

I'm away for the first part of the day (it's 8 AM here), but this afternoon I can put some commits together to complete Phase 3 (updating the keymaps).

Micro-Biology, 2018-12-16:
I'll make a pull request and that way you can see where I am up to, don't want replication of work.

[Dz60 layout consolidation #4671](https://github.com/qmk/qmk_firmware/pull/4671)

noroadsleft, 2018-12-16:
You raise a good point, @zvecr. The need to support Configurator users is not lost on me.

Realistically, I don't know that we're going to be able to strip the layouts down to the six @Micro-Biology has listed, because the [Community Layouts feature ](https://docs.qmk.fm/#/feature_layouts) needs three layouts on its own (and the DZ60 PCB supports others that aren't currently set up in the source code).

There are some layouts currently in the source that are just minor tweaks on other layouts. One possible solution I have is to add a default keymap JSON that's a standard 60% ANSI layout, but set to the LAYOUT_all macro, so someone can choose the DZ60 and hit the Load Default button and get something that works. Hopefully that will at least expose what goes where, if a user has extra keys.

zvecr, 2018-12-16:
For the Configurator heres some blue sky thinking, its almost like you need to model the options available, and make it a dynamic mapping from user selection to the Layout_all macro. However i understand that mapping the options against which are interdependent would be complicated.

[image]

Instead of picking an predefined macro, you would select 2 and 10, and the UI would reflect that suggestion. The user then picks their mapping and the magic translates this to the existing layout. That way its accessible to those who are comfortable with code and those who are not.

noroadsleft, 2018-12-16:
> For the Configurator heres some blue sky thinking, its almost like you need to model the options available, and make it a dynamic mapping from user selection to the Layout_all macro. However i understand that mapping the options against which are interdependent would be complicated.

That's actually what I'm thinking should be done. In fact, I had half a mind to suggest that be the standard for all boards, because once you figure out the "global" matrix and what positions share locations, you only have to set the matrix once and then point all the layouts at LAYOUT_all.

> Instead of picking an predefined macro, you would select 2 and 10, and the UI would reflect that suggestion. The user then picks their mapping and the magic translates this to the existing layout. That way its accessible to those who are comfortable with code and those who are not.

We actually have an issue against the Configurator talking about exactly that, but heck if I can find it right now. But the gist was to figure out a way to tell the Configurator what layout options are available (Split Space/Backspace/Shifts, ANSI/ISO, Standard/Tsangan/Winkeyless, etc.) by nesting JSON objects, similarly to the current structure. Then it's an issue of figuring how to relate them to each other.

noroadsleft, 2018-12-17:
@Micro-Biology wasn't sure where to drop this, but:

https://github.com/qmk/qmk_firmware/compare/master...noroadsleft:pull_4671

This incorporates your addition of the LAYOUT_60_c_iso macro, as well as some consolidation of the existing macros. It's branched from 398204b2a03bdac68fccb6f7592ed3ea52741ce5 as a base (the live master branch as I write this).

I was able to eliminate `LAYOUT_directional` by refactoring all the keymaps in the repo that used it. I think I could actually eliminate `LAYOUT_60_c_iso`, and fork `LAYOUT_directional` into ANSI and ISO variants. (I don't like either of those names, though.)

The more thought I put into this consolidation exercise, the more I start to think it's actually not a good solution.

As an aside, I'd like for @drashna and @mechmerlin to both weigh in on this, because they're the two people I've had the most discussion with on this topic.

Micro-Biology, 2018-12-17:
I didn't think about the configurator tbh, as I have never used it.

Is it possible to use several marcos for one layout? you could have a macro for each row of keys?

fshek, 2018-12-17:
I have soldered the DZ60 with this layout (support layout No. 2+3+6). I tried to use Layout_all, Layout, and all ISO layouts to map this key arrangement, but the the last key of row 4 (PgDn in my case) is always not responsive... It feels like the key is not present >.< 

I am a beginner of compiling firmware, and hope experts here could provide me some insights... 

[image]

Micro-Biology, 2018-12-17:
How are you programming your board? 

- If it is using the configurator, I don't think it is currently possible

- If however you are making your own hex file and flashing, for now you can use [my fork of qmk](https://github.com/Micro-Biology/qmk_firmware), and edit the iso_fullsplit layout to whatever you want your keys to be.

fshek, 2018-12-17:
I am not good at coding and I was using the configurator. Thanks for referring me to your fork. I think I need to boot up my ubuntu, set up the build environment, and try to compile myself during Xmas holiday xDDDD

noroadsleft, 2018-12-18:
@fshek This should get you close:

[dz60_fshek.json.txt](https://github.com/qmk/qmk_firmware/files/2688712/dz60_fshek.json.txt)

Save it, remove the `.txt`, go to Configurator, hit `Import Keymap`, give it that file. Should get you most of the way there.

The `LT()` you have next to right Alt, I have it as a `MO(1)` (same as left Windows key position), because I couldn't tell from your image what function you wanted on the `LT()`.

fshek, 2018-12-18:
Wow... Thank you very much. Will try your JSON file when I get back home tonight.

I have a special attempt on lower right hand side: 

|            PgUp |    RSFT_T(kc_up)   |  PgDn                 |
| LT(1, kc_left) | RGUI_T(kc_down) | RCTL_T(kc_right) |

Tap will be Up, Dn, Left, Right
Hold will be Shift, OS, Fn & Ctrl

I have tried this with "Layout_all" in the Configurator, and it is very handy for navigation. But the "PgDn" key assignment is not working on Configurator, that's why I throw out my question here. 

Hopefully I have some luck tonight with your JSON. Cheers

noroadsleft, 2018-12-18:
Your Shift/Up, GUI/Down and Ctrl/Right, are all directly supported in the Configurator, just drag the dual function keys where they go after loading the JSON, and then drag the tap functions (Up, Down and Right) onto the small boxes that appear on the keys. You'll see what I mean once you get to it. They're all under the "Mod key combinations" section.

For the `LT(1, KC_LEFT)`, drag the `ANY` key from the "QMK specific" section, and then type/paste `LT(1, KC_LEFT)` into the text box. It's a text input field like the `MO()` is.

I don't any reason on the QMK end why your Page Down key wouldn't work. That layout hasn't been changed in over three months, and it was confirmed as working at the time. I'm leaning toward that being a hardware issue on your end - bad switch, bad solder job, something like that.

fshek, 2018-12-18:
I have done a number of tests... 

PgUp is physically soldered in 12th location and PgDn is 14th location

When using LAYOUT_60_b_iso, both 12th (12th in key map) and 14th (14th in key map) keys are inactive.
[image]

When using LAYOUT_all, only 14th key (15th in key map) is gone. PgUp and Shift are working.
[image]
I assume LAYOUT_all should support maximum number of keys and should accommodate literally everything, but it does not turn out as expected.

Because the physical location of 12th key does not change but it has different behavior under 2 layouts, it drives me to think that it may be firmware issue.

You are right that it could be H/W or soldering issue. Just wanna test all possible HEX before trouble shooting the H/W.

noroadsleft, 2018-12-18:
Well, there are keymaps in the repo that use both those switch positions and zero issues have been reported there. One of the QMK Collaborators actually uses `LAYOUT_60_b_ansi` on his DZ60, and he would definitely report if there were a problem. This is why I'm fairly sure this is a hardware issue.

The key you have as Page Up is on the 12th column (electrically) in `LAYOUT_all`, and the 13th column in `LAYOUT_60_b_iso` and `_ansi`. It working on one spot and not the other definitely points me to hardware.

Have you tried physically swapping the switches to each other's places? You might be dealing with two different issues? (Forgive me if I'm telling you stuff you know; I'm trying to explore all avenues.)

fshek, 2018-12-19:
will check the switch and solder accordingly. Thanks for your help ;-)

closed 2019-11-21 by drashna