qmk/qmk_firmware #4683: Possible bug on process_leader.c

danielo515, 2018-12-19:
I think there is a bug on the current implementation of process_leader.c
Take a look at the snippet below (github should show you a preview of the code)

https://github.com/qmk/qmk_firmware/blob/584a7a03952dc5950b22fa4d035dc1c8ee98c87f/quantum/process_keycode/process_leader.c#L44-L51

As you can see, there is one check to only register the leader_time if LEADER_PER_KEY_TIMING is not enabled. However, just on the libe below that, you can see that it is executed anyway making no difference at all and executing the same code twice on certain situations.

Using the keyboard I didn't experienced any problem, so it may be irrelevant, but I wanted to point it out.

Regards

drashna, 2018-12-19:
It's not really a bug, but it's not optimal, yeah. 

This PR deals with that, and more: 
https://github.com/qmk/qmk_firmware/pull/4662

danielo515, 2019-02-22:
Fixed on the PR that @drashna pointed out.

closed 2019-02-22 by danielo515