qmk/qmk_firmware #4689: Bug with idobo: the whole first row sends multiple keycodes

cecton, 2018-12-20:
Short Story
=========

I have a Idobo keyboard ordered from Massdrop but I got a problem and the support couldn't help me so they refunded me (I didn't expect that... it's very nice but I still want my keyboard though).

I believe it's a problem in the PCB itself but before I buy another PCB I would like to have confirmation.

Issue Description
=============

There seems to be something wrong with the entire first row of the keyboard.

The rest is all working perfectly but when I press any key in the first row, instead of sending the one character, 4 (or 5) characters are sent, actually, all the characters of the column so if I press:

1 => 1qaz
2 => 2wsx
3 => 3edc
4 => 4rfv and a space (5 characters here because in the other columns the last key was a modifier)

The question is: is this a problem with the PCB or is it a problem with the software?

drashna, 2018-12-21:
Yeah, sounds like a diode issue, maybe?

cecton, 2018-12-21:
Well right now I see 2 options:
1. fixing it: but how do I find the failing diode?
2. buying another PCB: I thought about replacing it with the XD75RE. But does it have the exact same size and exact same?

mechmerlin, 2019-01-11:
This is a hardware issue and not a QMK issue. I personally would recommend just getting a new board. The idobo is marketed as a no soldering needed kit, the solution would be to desolder and then resolder diode(s) which defeats the purpose of no soldering lol. If this solution is satisfactory to you @cecton, can we close this issue?

cecton, 2019-01-11:
I didn't forget about this ticket, I'm waiting for the XD75RE to arrive to see if it works so I can update and close the ticket. It should arrive in a week or so.

cecton, 2019-01-16:
Solved!

So, as I suspected, the PCB of the XD75 is a perfect drop-in replacement for the idobo.

It fits in the case perfectly and the holes for the screws are located exactly at the same place.

closed 2019-01-16 by cecton