qmk/qmk_firmware #4692: Clueboard: third press of a MO key hangs firmware

khogeland, 2018-12-20:
The third key up event of a MO key causes the firmware to hang. I'm using [this](https://gist.github.com/khogeland/433b368460b6d16ef77787c7c0c23b1b) simple keymap, everything seems to work fine including the MO key/function layer. I've tried putting the MO mapping on other physical keys, so I don't think it's a hardware problem. Compiled at QMK firmware v0.6.209 (https://github.com/qmk/qmk_firmware/commit/f3cbb7bd6e307186e5802fa0366a21a5f9f5e41c) using the Docker build, same bug also present in the firmware built by the QMK configurator (v0.6.203).

khogeland, 2018-12-20:
6 presses on Caps Lock has the same effect

khogeland, 2018-12-20:
Noticed both keys trigger `led_set_kb` in `led.c`. Commenting out the body of that function prevents the issue, so something's broken in there. I don't even have LEDs, so for my personal use I'm going to leave that commented and call it fixed. :stuck_out_tongue:

drashna, 2018-12-21:
Do you know the last version that worked?

drashna, 2019-11-21:
Was this still happening?

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc