qmk/qmk_firmware #4731: Pin D5 for atmega32u4 not working with serial split

Legonut, 2018-12-27:
It seems that we can't use pin D5 with the common split code in serial. It's likely something to do with the pin being configured as an LED for the Pro Micro. This isn't an issue in I2C (the `lets_split_eh` uses it).

**System Information**
 - Keyboard:  Zygomorph (prototype)
https://github.com/Legonut/qmk_firmware/tree/zygomorph/keyboards/zygomorph
 - Operating System: Win10 w/ bash

 - avr-gcc version: 4.9.2

 - arm gcc version: 4.9.3

 - QMK Firmware version: 0.6.22

kageurufu, 2018-12-27:
I've broken this down to a minimal PoC

```
void matrix_init_kb(void) {
	matrix_init_user();

  DDRD |= _BV(PD3);
  PORTD |= _BV(PD3);
};
```

breaks pin atmega32u4 PD5 in the matrix.

drashna, 2018-12-27:
The serial part of the split common code doesn't support D5, actually.

https://github.com/qmk/qmk_firmware/blob/master/quantum/split_common/serial.c#L32-L65

Legonut, 2018-12-27:
@drashna we're not trying to use D5 as the serial pin, but as a column/row pin for the matrix

drashna, 2018-12-27:
Yeah, I realized that after. 

And ... unfortunately, I don't see where that's actually called by the split code, at all.

That-Canadian, 2018-12-27:
I'm looking into this too, it's very strange. Can confirm works fine with I2C or non-split boards as well.

@Legonut can you roll back your QMK to a commit before the serial code was changed?

I think this was the latest commit before the new serial code : https://github.com/qmk/qmk_firmware/commit/2149f3b5889f898fecde845a03d61c1a854d9393

drashna, 2018-12-27:
Looking at the code, there is noting that should "use" the D5 pin in the split code, serial or otherwise.  So I'm not sure why it's getting used. 

Though, I could be missing something obvious.

That-Canadian, 2018-12-27:
I agree with @drashna I'm at a loss. And the PoC that @kageurufu posted just makes me more confused haha. That *shouldn't* touch D5 at all...

That-Canadian, 2018-12-27:
Paging my buddy @nooges , I think he has more serial experience from me, maybe he can lend some insight?

drashna, 2018-12-27:
@kageurufu it may be worth converting those to the newer GPIO commands. 

Eg, I run this in the matrx init for my boards to turn off the LEDs on the Pro Micros. 

```c
    setPinOutput(D5);
    writePinHigh(D5);

    setPinOutput(B0);
    writePinHigh(B0);
```

While I used to use: 
```c
  DDRD &= ~(1<<5);
  PORTD &= ~(1<<5);

  DDRB &= ~(1<<0);
  PORTB &= ~(1<<0);
```

kageurufu, 2018-12-27:
Those compile to roughly the same C after macros are resolved

```c
// DDRD |= _BV(PD5);
 (*(volatile uint8_t *)((0x0A) + 0x20)) // 0x2A
      |= (1 << (5));
// setPinOutput(D5);
 (*(volatile uint8_t *)(( 0x00 + (((0x9 << 4) | 5) >> 4) + 1 ) + 0x20)) // 0x2A
      |=  (1 << (((0x9 << 4) | 5) & 0xF)); // (1 << 5)
```

I'm leaning towards this being an atmega32u4 quirk, not so much a QMK issue. I'm working on a test file in just avr-libc, without QMK to see

nooges, 2018-12-27:
D5 is the TX LED pin on the Pro Micro, so the code that touches that pin for LED may be interfering with serial comms.

kageurufu, 2018-12-27:
@nooges no code is touching that pin, I've disabled the TXLED macros on my machine for testing. 

And if you disable serial altogether, but enable D5 for output and set HIGH (see my test snippet, taken from `serial_master_init`) in the matrix_init, D3 stops responding

That-Canadian, 2018-12-27:
Digging through the Atmega docs now haha. It seems it might have something to do with USART1? I'm currently stumped, by reading through the datasheet now.

kageurufu, 2018-12-27:
I had that thought, I tried disabling USART1 (`USART1A = USART1B = USART1C = 0`) on matrix_init, and same effect

I have a debug key on my keymap to dump a ton of register values back out and confirmed it was all unchanged

nooges, 2018-12-28:
Looks like you'll have to add D5 info to this section: https://github.com/qmk/qmk_firmware/blob/master/quantum/split_common/serial.c#L32-L67 (noticed it while editing documentation today)

Legonut, 2018-12-28:
@nooges We're not trying to use it as the serial connect pin, just as a column/row pin.

nooges, 2018-12-28:
Oh I gotcha, missed that in the discussion. I've used D5 as a row pin just fine in my Iris Rev. 3 prototypes.

nooges, 2018-12-28:
My code for reference: https://github.com/qmk/qmk_firmware/blob/master/keyboards/iris/rev3/config.h

kageurufu, 2018-12-28:
Interesting. Can you try adding my D3 init from above (`DDRD |= _BV(PD3); PORTD |= _BV(PD3);`) somewhere and see if R5 still functions on your prototypes?

Legonut, 2018-12-30:
I tried the iris/rev3 code on my Zygomorph prototype, no luck on column 3. I only made the following changes:
`#define MATRIX_ROW_PINS { F1, F5, F7, B5, B4 }`
`#define MATRIX_COL_PINS { F4, F6, C7, D5, B6, D4 }`
`#define SOFT_SERIAL_PIN D3`

Legonut, 2019-01-10:
Mystery solved, was a crossed trace on my PCB (damned autorouter). Sorry about the mess.

closed 2019-01-10 by Legonut