qmk/qmk_firmware #4756: kernel-devel package on Fedora seems unnecessary

Edicatax, 2019-01-01:
**Describe the bug**
linux_install.sh installs the `kernel-devel` package as a dependency on Fedora.  This seems unnecessary. I ran through the first ten or so keyboards in `make all` and none of them failed because of a missing dependency.

**System Information**
 - Keyboard:  Sol
 - Operating System: Fedora 29
 - avr-gcc version: 7.2.0

 - arm gcc version: 7.1.0

 - QMK Firmware version: 0.6.218

 - Any keyboard related software installed? 
   - [ ] Auto Hot Key
   - [ ] Karabiner
   - [ ] Other

**Additional context**

germ, 2019-01-13:
Can you post the output of 'make all > build.log' , just to double check that this is a erroneous dependency?

Edicatax, 2019-01-13:
Here you go:
[build.log](https://github.com/qmk/qmk_firmware/files/2753476/build.log)

germ, 2019-01-14:
And just to double check, you don't have the devel package installed yeah?

Edicatax, 2019-01-15:
Turns out I had it installed when I ran that build, I removed it and reran the make all:
[build.log](https://github.com/qmk/qmk_firmware/files/2761579/build.log)

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc