qmk/qmk_firmware #4758: Tap dance doesn't properly handle multiple tap dance keys with modifiers held down

mnpqraven, 2019-01-01:
**Logic:**
A tap dance key for handling shift (`SFT_NM`) and a tap dance key for handling quotation mark (`QUOT`). Double tapping the quote key will send 2 quotation marks and a `KC_LEFT` to move the cursor inside the quote. It can also be used with shift to produce `"cursor in here"` but to avoid selecting the right quote when moving left, a unreg is required for shift then rereg it back after executing `KC_LEFT`.
 I'll be focusing on the hold option of the shift key and the double tap option of the quote key. 

Following are my code, i'll be excluding other cases (`TRIPLE_TAP` etc) for ease of reading, my complete tap dance setting can be found [here](https://github.com/mnpqraven/qmk_firmware/blob/master/users/othi/tap_dance.c)
`SFT_NM` [on press](https://github.com/mnpqraven/qmk_firmware/blob/master/users/othi/tap_dance.c#L181-L195) and [on release](https://github.com/mnpqraven/qmk_firmware/blob/master/users/othi/tap_dance.c#L211-L224) 
```
void dance_SFT_NM_finished (qk_tap_dance_state_t *state, void *user_data) {
    xtap_state.state = cur_dance(state);
    switch (xtap_state.state) {
        case SINGLE_HOLD:
            register_code (KC_LSFT);
#ifdef RGBLIGHT_ENABLE
            rgblight_sethsv_noeeprom_orange(); rgblight_mode_noeeprom(1);
#endif
            break;
```
```
void dance_SFT_NM_reset (qk_tap_dance_state_t *state, void *user_data) {
    switch (xtap_state.state) {
        case SINGLE_HOLD:
            unregister_code (KC_LSFT);
#ifdef RGBLIGHT_ENABLE
            rgblight_sethsv_noeeprom(OTHI_DEFAULT_R, OTHI_DEFAULT_G, OTHI_DEFAULT_B); rgblight_mode_noeeprom(OTHI_DEFAULT_MODE);
#endif
            break;
```

Quote key `QUOT` [on press](https://github.com/mnpqraven/qmk_firmware/blob/master/users/othi/tap_dance.c#L308-L319) and [on release](https://github.com/mnpqraven/qmk_firmware/blob/master/users/othi/tap_dance.c#L322-L327)
```
void dance_QUOT_finished(qk_tap_dance_state_t *state, void *user_data) {
    xtap_state.state = cur_dance(state);
    shift_pressed = get_mods() & ((MOD_BIT(KC_LSHIFT)|MOD_BIT(KC_RSHIFT)));
    switch (xtap_state.state) {
        case SINGLE_TAP: register_code(KC_QUOT); break;
        case DOUBLE_TAP: tap_code(KC_QUOT); tap_code(KC_QUOT);
                         if (shift_pressed) { //checks if Shift is held
                             //unregister_mods(MOD_BIT(KC_LSFT)); //this doesn't work, further testing required
                             //this is to remove the currently held mod key so pressing left won't select the right quote
                             clear_keyboard();
                         }
                         tap_code(KC_LEFT);
                         if (shift_pressed) {
                             //goes back to shifted as we are still holding down SFT_NM
                             register_code(KC_LSFT);
                         }
                         break;
    }
}
```
```
void dance_QUOT_reset(qk_tap_dance_state_t *state, void *user_data) {
    switch (xtap_state.state) {
        case SINGLE_TAP: unregister_code(KC_LSFT); unregister_code(KC_QUOT); break;
    }
    xtap_state.state = 0;
}
```

**Describe the bug**
**Testing**: `cursor` is the position the cursor will be on after executing a test

- single taps `QUOT`: 
     - Expect: 'cursor
     - Actual: 'cursor				**OK**
- double taps `QUOT`:
     - Expect: 'cursor'
     - Actual: 'cursor'				**OK**

- holds `SFT_NM`, single taps `QUOT`,releases `SFT_NM`, types `a`
     - Expect: "acursor
     - Actual: "a					**OK**

- holds `SFT_NM`, double taps `QUOT`,releases `SFT_NM`, types `a`
     - Expect: "acursor"
     - Actual: "Acursor"
in this case the shift modifier is still being held down after releasing `SFT_NM`, which should not happen because `dance_SFT_NM_reset` was executed(an indication to this is the RGB change, as when i release the `SFT_NM` key, the RGB changes back to `rgblight_sethsv_noeeprom(OTHI_DEFAULT_R, OTHI_DEFAULT_G, OTHI_DEFAULT_B);` and this was executed after `unregister_code(KC_LSFT);`

**System Information**
 - Keyboard:  
   - Revision (if applicable):  
 - Operating System: GNU/Linux
 - avr-gcc version: avr-gcc (GCC) 8.2.0

 - arm gcc version: arm-none-eabi-gcc (Arch Repository) 8.2.0

 - QMK Firmware version: `0.6.220`

closed 2019-01-06 by mnpqraven

weilbith, 2020-08-03:
@mnpqraven I think I have a similar issue. How did you solve it? You apparently just closed the issue...

mnpqraven, 2020-08-20:
I didn't, i decided to change the functionality of the key and decided to move shift tapdance to other keys