qmk/qmk_firmware #4766: Tada68 Unresponsive After Flashing

wadegbow, 2019-01-03:
**Describe the bug**
After updating my QMK repo to the master branch my Tada68 will not accept any of the compiled FLASH.bin files created from the default, or any users layouts. Transferring the FLASH.bin file like I always have, via cmd line, just results in an unresponsive keyboard.

Reverting back to the repo for QMK 0.5.41 will compile a working FLASH.bin, but I would like to try out some of the new user layouts and QMK features. Any help would be greatly appreciated.

**System Information**
 - Keyboard:  Tada68
 - Operating System: MacOS 10.13.6 (17G65)
 - avr-gcc version: avr-gcc (GCC) 7.3.0

 - arm gcc version: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

 - QMK Firmware version: 0.6.221

**Additional context**
Thinking it was something weird with my environment I got the vagrant box up and running and did a util/qmk_install.sh to compile it there and it's giving me the same outcome, an unresponsive keyboard.

tdgrunenwald, 2020-03-11:
I may have experienced the same issue recently. I thought I had nuked the bootloader and have been trying to ISP flash the QMK DFU bootloader. I may try to re-flash the mass storage bootloader to verify if I have the same issue

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc