qmk/qmk_firmware #4783: STM32 with OTG support has differently named USBEndpointConfig struct member

npdawson, 2019-01-05:
**Describe the bug**
STM32 chips with USB OTG support apparently use the OTGv1 HAL code instead of the USBv1 code. In hal_usb_lld.h for OTGv1, the USBEndpointConfig struct has an in_multiplier field:
```c
/**
 * @brief   Determines the space allocated for the TXFIFO as multiples of
 *          the packet size (@p in_maxsize). Note that zero is interpreted
 *          as one for simplicity and robustness.
 */
uint16_t in_multiplier;
```
instead of USBv1's ep_buffers field:
```c
/**
 * @brief   Reserved field, not currently used.
 * @note    Initialize this field to 1 in order to be forward compatible.
 */
uint16_t ep_buffers;
```

The problem occurs in tmk_core/protocol/chibios/usb_main.c where the QMK_USB_DRIVER_CONFIG macro defines 3 USBEndpointConfigs, all with `.ep_buffers = 2,`. When compiling for my STM32F4 Discovery board, I get an error saying that USBEndpointConfig has no member named ep_buffers.

**System Information**
 - Keyboard:  STM32F4 Discovery board
   - Revision (if applicable):  
 - Operating System: Linux
 - avr-gcc version: 8.2.0

 - arm gcc version: 8.2.0

 - QMK Firmware version: 0.6.222

 - Any keyboard related software installed? 
   - [ ] Auto Hot Key
   - [ ] Karabiner
   - [ ] Other

**Additional context**

yiancar, 2019-01-05:
Could you possibly use the ifdef for stm32f4xx to modify the endpoint defines?

npdawson, 2019-01-06:
Yes, it works for my discovery board and compilation is successful for planck rev6.

awkannan1, 2019-01-11:
I will try to check it on my STMF072 discovery and a blue pill this weekend

chipperdoodles, 2019-07-01:
I'm trying to compile this for my nucleo stm32f401re, If i set console enable = yes in rules.mk the build fails, not sure if that's just a gimme. It compiles ok with console enable = no. I'm not sure i've configured my build for this mcu right either, if anyone has some examples of a setup for an stm32f4xx I can try testing more!

PR #7564 (2019-12-14): Fix ChibiOS USB config for chips which support OTGv1

## Description

As per #4783 USBv1 devices have `ep_buffers` and OTGv1 have `in_multiplier`.

This PR aligns usb endpoint config as per rest of file, using explicit order instead of struct member name.

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #4783

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requ [... cut]
commit 80c2e267 (2019-12-14): Fix ChibiOS USB config for chips which support OTGv1 (#7564)

* Align endpoint config as per rest of file (fixes #4783)

* Add comments about explicit order use

* Update tmk_core/protocol/chibios/usb_main.c
closed 2019-12-14 by zvecr