qmk/qmk_firmware #4785: Switching between left and "right" shift doesn't keep the shift modifier enabled

rayburgemeestre, 2019-01-05:
Didn't file this as a bug because it could be I'm not reading the docs correctly.

I have this as my left shift: `KC_LSFT` and `SFT_T(KC_ENT)` as my "right" shift.
It's really super convenient.

The problem is that I tend to switch back and forth between left- and right shift when trying to type something IN ALL CAPS. 

An example would be typing WATERLOO.
I tend to use the right-shift for WATER, then I switch to left-shift for the LOO type (on a qwerty layout). It's more convenient for my fingers and I do it automatically. However when I do that, I will get WATERloo, because `SFT_T(KC_ENT)` doesn't seem to take over the role that I started with the key-down.

Hope it makes sense, would really appreciate a solution / workaround for this it would make the entire thing perfect for me :)

shmcgrath, 2019-01-28:
Try swapping to the RSFT_T(EC_ENT) instead of SFT_T(KC_ENT). I don't know if it will have an impact, but right now, you are calling left-shift twice

rayburgemeestre, 2019-01-29:
Wow, thanks a lot! Really! Just tried your suggestion and _finally_ your suggestion fixed my issue. Now I can unlearn some of the muscle memory I developed working around it :)

Glad that there is no problem with QMK but indeed with my configuration (as I expected)
Thanks again!

closed 2019-01-30 by rayburgemeestre