qmk/qmk_firmware #4801: Macros send random characters

luandy64, 2019-01-08:
I'd like someone to replicate my issue and prove my solution works. Then we can change the docs accordingly.

### Problem: Following the docs to define a macro, I write the following code:
```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case MY_MACRO:
            if (record->event.pressed){
                SEND_STRING("Hello World");
            }
        break;
    }
    return true;
}
```

When I go to use `MY_MACRO`, I get "Hello World" and some other character at the end (Multiple tests have given me multiple things, like a `+` or `:` or `|`).

### Solution:
```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case MY_MACRO:
            if (record->event.pressed){
                SEND_STRING("Hello World");
            }
        return false;
        break;
    }
    return true;
}
```

Now the macro works as intended.

I question whether we need both the `return false;` and the `break`. I didn't look into and I am not confident in my C skills to make a confident statement about this. But I have `return false; break;` in my keymap right now and it's working.

vomindoraan, 2019-01-08:
Could you link to where exactly in the docs you found this? The `return false;` is indeed required. The `break;` isn't as the `return` statement already serves to break the switch.

luandy64, 2019-01-09:
[Here's the link](https://docs.qmk.fm/#/feature_macros?id=the-new-way-send_string-amp-process_record_user)

vomindoraan, 2019-01-09:
While the tutorial is technically correct and `return false` isn't required since the macro keycode is `>= SAFE_RANGE`, it'd still be nice to have it there. In your case, how did you define the `MY_MACRO` keycode?

luandy64, 2019-01-09:
Just in an `enum` with a bunch of other things.
```
enum custom_keycodes {
    QWERTY = SAFE_RANGE,
    ...
    MY_MACRO
};
```

Is this the proper way of doing it?

vomindoraan, 2019-01-10:
It is, but then it shouldn't have caused any issues. Could you share the whole enum?

luandy64, 2019-01-10:
```
enum preonic_keycodes {
  QWERTY = SAFE_RANGE,
  LOWER,
  RAISE,
  TMUX_LD = LCTL(KC_RBRC),
  TMUX_SC, // TMUX Scroll
  TMUX_LO, // TMUX LogOut
};
```
where `MY_MACRO` is `TMUX_LO`

luandy64, 2019-01-11:
Thanks for all of your help @vomindoraan!

closed 2019-01-11 by luandy64