qmk/qmk_firmware #4816: Diverge3 broken by #4419

mcmadhatter, 2019-01-10:
The soft serial stuff for issue #4419 has stopped the diverge 3 from working. Diverge 3 config.h will need a slight rework to remove the soft serial and make it I2C based again.

mtei, 2019-01-11:
You can try adding the following options:

```
diff --git a/keyboards/diverge3/config.h b/keyboards/diverge3/config.h
index 464dc835f..29e2edfba 100644
--- a/keyboards/diverge3/config.h
+++ b/keyboards/diverge3/config.h
@@ -59,6 +59,7 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 
 /* serial.c configuration for split keyboard */
 #define SOFT_SERIAL_PIN D0
+#define SELECT_SOFT_SERIAL_SPEED 3 /* 1:default, 2, 3, 4, 5 */
 
 /* define if matrix has ghost (lacks anti-ghosting diodes) */
 //#define MATRIX_HAS_GHOST
```

mcmadhatter, 2019-01-11:
It should not be using serial , I think just changing over to use #define  USE_I2C instead should be enough but I haven’t had time to try it out as a fix yet.

mcmadhatter, 2019-01-11:
setting the speed to 2 seems to work ok.

drashna, 2019-02-08:
@mcmadhatter Are you using the 8MHz version of the Pro Micro?  If so, that may be the issue. 

And just curious, did 2 or 1 work, at all?  Or only 3 (and up)?

mcmadhatter, 2019-02-26:
16mhz version,  2 worked fine for me so I stuck with that.

PR #5076 (2019-08-23): [Keyboard] Add Soft Serial Speed to Diverge3 keyboard config
## Description
Set the `SELECT_SOFT_SERIAL_SPEED` setting for the Diverge 3, since it seems to have issues with the newer split serial code. 

## Types of changes

- [ ] Core
- [ ] Bugfix
- [ ] New Feature
- [ ] Enhancement/Optimization
- [X] Keyboard (addition or update)
- [ ] Keymap/Layout/Userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by this PR

* Fixes #4816 

## Checklist:

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have up [... cut]
closed 2019-04-23 by mcmadhatter