qmk/qmk_firmware #4838: Command feature not working since single HID endpoint was introduced

vomindoraan, 2019-01-13:
**Describe the bug**
The [Command](https://docs.qmk.fm/#/feature_command) feature is not working. It appears that the single endpoint PR (#3951) broke it. Replicated on AVR and ChibiOS (Kiibohd) boards.

When holding down the trigger combination for Command (default: LShift+RShift) and pressing one of the magic keys (e.g. D for debug mode), the key gets passed straight to the host without triggering the command. On versions prior to the aforementioned HID endpoint change, the keyboard would activate debug mode correctly and would never send the D key to the host machine.

**System Information**
 - Keyboard: KBD6X, Melody96, Nightfox
 - Operating System: Windows (MSYS2), Docker
 - avr-gcc version: 5.4.0 and whatever [`qmkfm/qmk_firmware`](https://hub.docker.com/r/qmkfm/qmk_firmware/) uses
 - arm gcc version: 6.3.1 and whatever [`qmkfm/qmk_firmware`](https://hub.docker.com/r/qmkfm/qmk_firmware/) uses
 - QMK Firmware version: 0.6.231
 - Any keyboard related software installed? 
   - [ ] Auto Hot Key
   - [ ] Karabiner
   - [x] Other: WinCompose, KbdEdit

vomindoraan, 2019-01-14:
@abrasive Do you have any idea as to which part of your PR might have caused this?

fauxpark, 2019-01-14:
Weird... I have no issues with it here. I've got a Satan with basically default rules.mk except for:

```make
#CONSOLE_ENABLE = no
#COMMAND_ENABLE = no
RGBLIGHT_ENABLE = no
KEYBOARD_SHARED_EP = yes
```

I get this in the toolbox:

```
*** SATAN: GH60 Akuma connected  -- F055:60AA:0003 (8&314c9ff2&0&0000)
  > 
    	- Version -
    DESC: QMK keyboard firmware for Satan GH60 with WS2812 support
    VID: 0xF055(SATAN) PID: 0x60AA(GH60 Akuma) VER: 0x0003
    BUILD: "929065" (00:34:18 Jan 15 2019)
    OPTIONS: LUFA EXTRAKEY CONSOLE COMMAND NKRO 4096
    GCC: 4.9.2 AVR-LIBC: 2.0.0 AVR_ARCH: avr5
```

Might be at the keyboard level?

vomindoraan, 2019-01-14:
Here's how it behaves for me on Whitefox: https://i.imgur.com/VYkbX8e.gif
Same for the other two boards I tried it with (KBD6X and Melody96).

vomindoraan, 2019-01-14:
I just tried it with `KEYBOARD_SHARED_EP = yes`, and it does work. Now to get to the bottom of why that is.

fauxpark, 2019-01-14:
What's with the tab? Also, what happens when you hit C? (I believe that actually enables the console printing)

vomindoraan, 2019-01-14:
It prints fine without C when it's working (see [here](https://i.imgur.com/zhtY8tp.gif)).

vomindoraan, 2019-01-15:
Can't we just use
```c
#define IS_COMMAND() (get_mods() == (MOD_BIT(KC_LSFT) | MOD_BIT(KC_RSFT)))
```
instead? I'm not sure why `keyboard_report->mods` was used in the first place. The only difference would be that weak mods and oneshot mods aren't accounted for in `get_mods()`, but I don't think that's much of a problem.

fauxpark, 2019-01-15:
Oh, that's much nicer, I didn't even realise that was a thing.

As for the oneshot/weak mods, you could just `|` them together. Along with `get_mods()` there is also `get_weak_mods()`, `get_macro_mods()`, and `get_oneshot_mods()`.

vomindoraan, 2019-01-15:
Indeed. I suspect that might have been the reason why `keyboard_report->mods` was chosen in the first place: it's exactly all of those OR'd together (see [here](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/action_util.c#L140-L152)).

vomindoraan, 2019-01-15:
I think checking just the real mods with `get_mods()` should be fine. It will prevent people who use one shot mods from activating commands like so:
```
LShift↓ LShift↑ RShift↓ RShift↑ key↓ key↑
```
But they should still be able to do it by holding LShift+RShift and pressing the command key as you normally would. In other words,
```
LShift↓ RShift↓ key↓ key↑ RShift↑ LShift↑
```
should work even if LShift and RShift are set up as one shot mods. I'll need to test this to confirm.

fauxpark, 2019-01-15:
Well I guess once #4301 is merged we can go about changing the remaining `keyboard_report->mods` to `get_mods()`. It should be a much easier job then 😁

vomindoraan, 2019-01-15:
My thoughts exactly 😃
I will test the one shot mods scenario later today and report back if I find any issues.

patrickmt, 2019-01-24:
Since the arm_atsam code runs mostly its own USB implementation, I do see misalignment issues when `KEYBOARD_SHARED_EP` is enabled. NKRO and MOUSE sharing are already being forced undefined, so I believe the best thing to do for now is to
```
#if defined(PROTOCOL_ARM_ATSAM)
  #undef KEYBOARD_SHARED_EP
#endif
```
in tmk_core/common/report.h around line 79, before `KEYBOARD_REPORT_SIZE` gets modified.

I've also confirmed changing `keyboard_report->mods` to `get_mods()` works properly.

drashna, 2019-01-24:
I can add the `get_mods()` update to my IS_COMMAND PR, right now, as that would fix it in one go. Especially, since I'm already touching so many files....

drashna, 2019-01-25:
Sounds good.

vomindoraan, 2019-01-27:
PR opened here: #4955