qmk/qmk_firmware #4851: speed for rgb underglow

dragonlost, 2019-01-15:
I programmed my gh60 with the RGB_SPI and RGB_SPD but it does not seem to work.

Is it me that is not understood something or is it a bug?

dragonlost, 2019-01-16:
I just realized that the BL_TOGG key does not work either. But BL_INC and BL_DEC work.

mechmerlin, 2019-01-16:
The GH60 does not come with RGB underglow support by default. There are no LEDs connected to mcu pins. The `RGB_` keycodes will not work.

Funny that `BL_INC` and 'BL_DEC` work though because that's not turned on in QMK by default either. However the keyboard does physically support these. You can look at the GH60's `rules.mk` to turn the options on.

dragonlost, 2019-01-16:
Sorry i got it wrong my keyboard is an XD60 exactly

drashna, 2019-02-08:
@daisukekoba The RGB Speed options are only for the RGB matrix feature supports this.  This keycode does not support the RGB Light (underglow) feature.

So, it would do nothing for you, based on the hardware (XD60 or GH60).

dragonlost, 2019-02-08:
yet using the keys "rgb mode +" and "rgb mode -" I arrive after a while found a faster mode.

drashna, 2019-02-08:
Correct. This because it toggles through the modes, and the modes have faster versions. 

https://docs.qmk.fm/#/feature_rgblight?id=keycodes

So, this is a discrepancy between the two modes.

closed 2019-11-21 by drashna