qmk/qmk_firmware #4879: LCTL(KC_C) and LCTL(KC_V)

Breganov, 2019-01-18:
I've made two buttons with said keys functions in the title: 
* LCTL(KC_C);
* LCTL(KC_V).

Tapping on them more then once with time period less then 200 ms makes my keyboard reload. It began doing so after RGB update on ErgoDox Ez configurator. Later versions of .hex files working well with said key functions and they don't reboot the whole keyboard. 

Please help. I don't understand what's going on in the update. So I can't figure out where to make changes.

Thank you in advance.

drashna, 2019-02-08:
If you're compiling from source, set `COMMAND_ENABLE = no` in your rules.mk, and recompile.

Breganov, 2019-02-08:
It didn't help. I've made it `COMMAND_ENABLE = no`. But it still reboots each time. 
I've tried to switch off RGB lights and Swaping Hands function. But it still reboots each time I press LCTL(KC_C) or LCTL(KC_V) keys. And later versions of compiled .hex files work fine for me with this key function. I'm not sure where to search.

Breganov, 2019-02-12:
I've found out that there is `C(kc)` command that does the same thing. While `KC_COPY` and `KC_PASTE` doesn't work for me somehow on my Windows 10 PC I've compiled this one. And I've got the same problem with QMK Firmware 0.6.272.

`C(KC_C)` and `C(KC_V)` reboot my keyboard. And for a short period of time, it becomes unavailable for input. I've made two versions of .hex files. One with `COMMAND_ENABLE = yes` and the other `COMMAND_ENABLE = no`. Neither `yes` nor `no` version fixed my problem.

Breganov, 2019-02-13:
I've found new thing. The said commands work incorrectly only on my right side of my keyboard. And somehow the only new one .hex file. The old .hex files work perfectly well.

Still researching the problem.

drashna, 2019-04-01:
Was this still an issue? 

And can you post a link to the Ergodox EZ profile that you're using?

Breganov, 2019-04-03:
Hi, @drashna, thank you for your response.

> Was this still an issue?

Still it is an issue. I can't figure it out why it's always blinking the leds and doesn't respond to other keys while it's blinking. I've tried my old compiled .hex. It was perfectly fine. 

But at some point, and I didn't find it, keyboard started to act weird. 

> And can you post a link to the Ergodox EZ profile that you're using?

I can't. I've downloaded MSYS and qmk_firmware to be able to compile it myself. But issue still exists. ATM I gave up and using standart `CTRL` + `C` combinations on key board. It works fine. `C(KC_C)` doesn't work properly anyways. 

I found out that some functions act weird on my left hand. But are fine on the right hand. It was with `LCTL(KC_PSCR)` for making screenshots faster. And now I'm guessing why?

But I'm getting the same strange LED blinks on a combintaion of `ALT_T(KC_DOWN)` and `KC_TAB` when I'm jumping over the applications on my Windows and pushing them rapidly. Those keys are on my left hand. And I'm not getting the same LED blinks when I'm using my lefthand `CTL_T(KC_LEFT)` + `KC_C`.

I can place here my keymap.c file if you want to.

drashna, 2019-04-05:
Could you post your keymap.c file (post it on gist.github.com and link it here)?

drashna, 2019-04-05:
> I found out that some functions act weird on my left hand. But are fine on the right hand. It was with `LCTL(KC_PSCR)` for making screenshots faster. And now I'm guessing why?

If that is the case, it may be an issue with your TRRS cable.  Or the IO Expander in the left side.
The right side has the MCU in it, and is directly connected to that.  The left side uses an expander, and is connected over the TRRS cable. 

It could be that one of these is faulty, and that could absolutely cause issues.

drashna, 2019-07-23:
Actually, just reviewing this, this may be because the `i2c_master` code was being optimizing when compiling, which can lead to issues.   That's been fixed now and may help with this issue. 

For now, I'm going to close this issue.  If this is still a problem, then please re-open the issue

closed 2019-07-23 by drashna