qmk/qmk_firmware #4895: Cannot enter DFU mode with whitefox's reset key

kissu, 2019-01-20:
Hi ! :hand: 
I'm a bit new to QMK but after some research on wiki, I cannot find a way to make the `RESET` button work.
Compilation and flashing with physical button on the back are totally fine ofc, just not as convenient as a button press.

**System Information**
 - Keyboard: `Whitefox` :fox_face: 
 - Operating System: `Ubuntu 18.04`
 - avr-gcc version: `5.4.0`
 - arm gcc version: `6.3.1 20170620`
 - QMK Firmware version: `0.6.212`

**Additional context**

Since I'm not sure how to explain things better, I have a screen of the whole thing, if it can help somehow...
[image]

I found the [AVR MCU options page](https://beta.docs.qmk.fm/reference/config_options#avr-mcu-options), but I don't know what can I use there since, in my case, I do have a `KIIBOHD_BOOTLOADER` AFAIK (in my root `whitefox` directory).
For the reset, I tried the `RESET` keycode and that little snippet
```c
void safe_reset(qk_tap_dance_state_t *state, void *user_data){
  if (state->count >= 3){ // Reset the keyboard if you tap the key more than three times
    reset_keyboard();
    reset_tap_dance(state);
  }
}
```
both produce the same result ofc.

drashna, 2019-02-08:
Do you know when this started?  Eg, when it was working?  

Also, you can use `dfu-util-wait` as the target, to do the waiting part automatically.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc