qmk/qmk_firmware #4901: Erratic LED behavior on Ergodox

dpapavas, 2019-01-21:
I'm having trouble similar to what is described in #2471 with a DIY Ergodox, on Linux (Debian Stretch).  Apart from the similar symptoms, the problem seems to have the same root cause (incorrect interpretation of HID reports) and it goes away if I set `NKRO_ENABLE = no`.

Unfortunately, the introduced fix (#4824), doesn't seem to work for me.  It seems though, that removing the ` || report_id == REPORT_ID_NKRO` clause [here](https://github.com/tomwang2011/qmk_firmware/blob/c8338a5509d0ac263d330b4454443913ce61f8bf/tmk_core/protocol/lufa/lufa.c#L528) (introduced in commit  39bd760faf2666e91d6dc5b199f02fa3206c6acd) fixes the problem in my case, without breaking anything, as far as I can tell.

I'm creating this as a bug report, instead of a PR, because I have no idea whether that's an appropriate fix (it probably isn't, as it was based on guesswork).  Hopefully someone who is more familiar with the HID protocol, will have a better idea.

fauxpark, 2019-01-21:
Could you try removing the if statement around `report_id = Endpoint_Read_8();`? It actually doesn't make sense to have that if we've already established there are two bytes in the endpoint, otherwise it could potentially only read the first byte and think the report ID is the LED state.

fauxpark, 2019-01-22:
I spun up a Debian Stretch VM and although that if statement shouldn't be there (and I don't have an Ergodox), I can't seem to reproduce this. The only thing I can think of is that your repo is not actually up to date.

dpapavas, 2019-01-22:
Thanks for your quick response fauxpark.  I tried the following:  I checked out a fresh copy of the current master branch and applied the following changes:

```diff
diff --git a/tmk_core/protocol/lufa/lufa.c b/tmk_core/protocol/lufa/lufa.c
index cdabaf16e..bbbb9931b 100644
--- a/tmk_core/protocol/lufa/lufa.c
+++ b/tmk_core/protocol/lufa/lufa.c
@@ -521,15 +521,15 @@ void EVENT_USB_Device_ControlRequest(void)
                     if (Endpoint_BytesInEndpoint() == 2) {
                       uint8_t report_id = REPORT_ID_KEYBOARD;
 
-                      if (keyboard_protocol) {
-                        report_id = Endpoint_Read_8();
-                      }
+                      report_id = Endpoint_Read_8();
 
                       if (report_id == REPORT_ID_KEYBOARD || report_id == REPORT_ID_NKRO) {
                         keyboard_led_stats = Endpoint_Read_8();
+                        uprintf("report_id: %d, leds: %d\n", report_id, keyboard_led_stats);
                       }
                     } else {
                       keyboard_led_stats = Endpoint_Read_8();
+                      uprintf("leds: %d\n", keyboard_led_stats);
                     }
 
                     Endpoint_ClearOUT();
```

I also added a custom keymap with the following:

```c
void led_set_user(uint8_t usb_led) {
    if (IS_LED_ON(usb_led, USB_LED_CAPS_LOCK)) {
        ergodox_right_led_2_on();
    } else {
        ergodox_right_led_2_off();
    }

    if (IS_LED_ON(usb_led, USB_LED_SCROLL_LOCK)) {
        ergodox_right_led_3_on();
    } else {
        ergodox_right_led_3_off();
    }
}
```

The map also has a `rules.mk` file with `CONSOLE_ENABLE = yes`.  The problem persists.  When I switch language (which is configured to turn on the scroll lock led) and the press caps lock, I get the follwoing output:

```
leds: 4
leds: 6
report_id: 5, leds: 2
```

The first line is emitted after I change language and the scroll lock lights up.  Then, when I press caps lock the other two lines are printed out.  The leds are both on for a split second after I hit the caps lock, but then the 2-byte report arrives and the second byte doesn't contain the scroll lock bit.

Any idea what might be causing this?

fauxpark, 2019-01-22:
I think I understand now. Linux seems to treat the 6KRO and NKRO interfaces as separate keyboards, even though they're part of the same device. Like Windows, it shares the LED state across keyboards, but evidently the language switching stuff is only per keyboard. This is probably expected behaviour.

What we actually need to do is check whether NKRO is currently on and discard one of the reports based on that and the report ID (if any). I'll look into it, after I get some sleep :)

dpapavas, 2019-01-22:
Your hypothesis appears to be correct.  I get two entries with name `ErgoDox EZ ErgoDox EZ` in `/proc/bus/input/devices` (although their difference is not very clear) and when I set the group (language) toggle LED to caps lock (which is the only LED the keyboard of the laptop that the EZ is attached to has) the caps lock LED on the laptop keyboard does not light up when I switch language, unlike the one on the EZ, which does.  When the caps lock LED has the usual function, both are in sync, as you say.

Thanks again for looking into this and let me know if I can help in any way.

drashna, 2019-02-08:
@fauxpark This was fixed in the core code, right? 

I want to say it was, but I'm not 100% sure.

drashna, 2019-02-08:
I checked, and yes, it looks like it was.

closed 2019-02-08 by drashna

fauxpark, 2019-02-08:
@drashna - This is a slightly different issue - Linux apparently sees the 6KRO and NKRO interfaces as two different keyboards, and while it does sync caps/num/scroll lock led state between them, the functionality that turns on the scroll lock LED when the language is switched doesn't.

[Here](https://gist.github.com/fauxpark/eaa298311dbb2be2cb033abf441fe403) is *kind of* a solution (untested, which is why I haven't gotten back to this yet - sorry), but I suspect that if you toggle NKRO this language switching scroll lock LED will become out of sync. There's not much that can be done in that case. AFAIK there is no way for the keyboard to ask for the LED state.

dpapavas, 2019-02-11:
I had resorted to disabling the NKRO feature, as I don't really need it (nor had I ever enabled it), but I tried to compile and test the diff.  It seems to work fine.  I've verified that I actually toggle NKRO, by using `xev` to see whether I can send more than 6 keypresses at the same time and the leds seemed to always stay in sync.  You might want to test for yourself though, as I might be missing something here.