qmk/qmk_firmware #4925: Add MIDI-CI Support

brandenbyers, 2019-01-24:
# Add MIDI Capability Inquiry (MIDI-CI) Support

## Feature Request Type

- [x] Core Functionality
- [ ] Add-on hardware support (e.g. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing Feature(s)
- [ ] New behavior

## Description 

Given QMK supports MIDI, are there any plans to implement the MIDI-CI spec? If there are no plans, is QMK open to it being added? For reference, see the [MIDI-CI spec](https://www.midi.org/articles/midi-manufacturers-association-mma-adopts-midi-capability-inquiry-midi-ci-specification) (sorry I can't post a direct link; the spec is behind a free user signup).

With the Proton C and rotary encoder support, QMK is positioned perfectly for an advanced MIDI controller with two-way communication.

brandenbyers, 2019-01-24:
That's good to know for AVRs, but what about ARM support? Does QMK also lean heavily on LUFA for ARM?

Personally, I'm mostly interested in ARM support. But I'll also post a request on LUFA so as to hopefully keep MIDI feature parity between ARM and AVR.

fauxpark, 2019-01-24:
LUFA is only designed for AVRs and AFAIK is the only way to do MIDI in QMK at the moment. Some ARM boards (I think STM32?) run LUFA on top of ChibiOS, though, which is how the latest Plancks can do MIDI stuff.

It might also be worth noting that the existing MIDI support in QMK is largely WIP/untested.

brandenbyers, 2019-01-24:
Ok, now this is starting to make more sense! LUFA on top of ChibiOS is why I'm able to use MIDI with Proton Cs.

MIDI support in QMK isn't fully untested. I have been testing it off-and-on for my own prototypes for the past two years! 😄

I'd love to contribute native support for MIDI on ARM boards, although I'll be quite slow if I can pull it off. Like I said before, I think the Proton C + MIDI-CI (and in the future MIDI 2.0) would be great!

brandenbyers, 2019-01-24:
First step, [cross posted to LUFA issues](https://github.com/abcminiuser/lufa/issues/144) to inquire about getting MIDI-CC added to LUFA.

fauxpark, 2019-01-24:
> MIDI support in QMK isn't fully untested. I have been testing it off-and-on for my own prototypes for the past two years! 😄

Ah, well the (rather lacking) documentation says so in at least [two](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_audio.md#midi-functionality) [places](https://github.com/qmk/qmk_firmware/blob/master/docs/getting_started_make_guide.md#rulesmk-options) :P

koshimazaki, 2019-09-25:
@brandenbyers i would be more then happy to help with testing. Do you by any chance have any keymaps with use of midi that you are happy to share?  Especially the ones beyond use of keycodes as notes. For example 
`publicvoidmidi_send_programchange(MidiDevice* device,uint8_t chan,uint8_t num)`

brandenbyers, 2019-09-25:
I haven’t used MIDI program change on a QMK board before, but I’d be happy to experiment with it if it is something that you could use and you’re interested in helping test out and improve QMK MIDI support.

If so, it will probably be best if we create a new Github issue to improve MIDI documentation and/or implementation to track progress on this given this issue is specific to MIDI-CI. Let me know if that is of interest and I’ll open a new issue.

koshimazaki, 2019-09-25:
Yes thats awesome lets do that. Thank you! I gave it as example, looks like the functions will be similar to CC msges as well or pitchbend etc. I have good use case for program change tho.

koshimazaki, 2019-10-06:
@brandenbyers please let me know if you had any luck with this. I am testing recent firmware and can't get any midi appear in Max/MSP. I can see Preonic (V3) as midi device, can send midi to it and sequence internal synth but cant send any notes or midi information using it. It doesnt work as midi controller basically. Tried enabling MIDI with keycodes and using MIDI note keycodes as well. Wonder if there are any `#includes` missing in my keymaps. Enabled MIDI in my rules file.
@fauxpark @drashna perhaps you might know?

brandenbyers, 2019-10-06:
The latest example MIDI layout that I did was for the Dora: https://github.com/qmk/qmk_firmware/blob/master/keyboards/ckeys/thedora/keymaps/default/keymap.c

brandenbyers, 2019-10-06:
For a more advanced usage of MIDI, I also did the MIDIany, but just realized I never shared that back upstream so you’ll have to go to the c-keys fork for that: https://github.com/c-keys/midi-any

koshimazaki, 2019-10-06:
Thanks for the links @brandenbyers i will investigate these and get back to you.

koshimazaki, 2019-10-06:
I tested it and notes and keycodes like `MI_E_3` are working fine with default setup. Problem is once MIDI is enabled it changes entire keyboard to MIDI keyboard. How about having only few keys being MIDI notes? Sort of hybrid setup. What I would like to do is have layer for sending programme change msges and have only few, say 1-5 keys being responsible for sending programme change. Is that possible at all? I think would be great to have example code for that for future use. 

Also, once we are at it, I realised there is no info on Preonic v3 needing extra packages for chibios. Keymap doesn't compile. Might be worth adding it in the main page for preonic v3 and likely for Planckv6 too.

closed 2020-02-03 by stale[bot]

jonmccon, 2020-05-20:
@koshimazaki Just ran down this path trying to find out why I couldn't get midi support on a proton c to compile. Have you been able to find any progress on midi controllers via qmk? 
I'd love to help test this as well, I'm looking to build a controller with both normal keys & rotary encoders on i2c, But when testing on a BDN9 I'm running into an error chibios essentially missing. 
I'll check out the keymaps you linked here and can report back. thanks!

koshimazaki, 2020-05-20:
@jonmccon yeah i managed to solve my issue which was user error of not being able to code properly lol and see what to put in the commands ;) Anyway, solution for my use case was defining custom keys:
  `'case` _PATCH1:
                  if (record->event.pressed) {  
                      midi_send_programchange(&midi_device, 15, 0);

                  }
                  return false; `break;`
Here the channel is 16 and program change number is 1. It counts from 0.
I include the following libraries in my keymap. Need to upload it to github still. 
#include "muse.h"
#include "preonic.h"
#include "midi.h"
#include "qmk_midi.h"
I think the last two are most important for other keyboards then preonic. 

i havent solved i2c stuff yet. I want to use my keyboards with [ER301 ](http://www.orthogonaldevices.com/er-301)so its quite specific.

jonmccon, 2020-05-21:
I have everything setup like your examples but I'm continuing to get this ChibiOS error while compiling as soon as I turn on `MIDI_ENABLE = yes`

```
Compiling: tmk_core/protocol/chibios/main.c                                                        In file included from tmk_core/protocol/chibios/main.c:51:
tmk_core/protocol/midi/qmk_midi.h:7:36: error: unknown type name 'MIDI_EventPacket_t'
 void              send_midi_packet(MIDI_EventPacket_t* event);
                                    ^~~~~~~~~~~~~~~~~~
tmk_core/protocol/midi/qmk_midi.h:8:36: error: unknown type name 'MIDI_EventPacket_t'
 bool              recv_midi_packet(MIDI_EventPacket_t* const event);
                                    ^~~~~~~~~~~~~~~~~~
```

Is there a library or function type I would be missing?

koshimazaki, 2020-05-21:
O think you might need to update core libraries like chibios. Are you on newst qmk? I had to update libraries separaty.

alexfry, 2021-01-11:
I would also find this useful, specifically for changing the colour of LEDs on my board (that sit next to my encoders) over MIDI. 

I've done this in the past with the "DJ Tech Tools MIDI Fighter Twister"