qmk/qmk_firmware #4930: Script to create index of keyboard supported by QMK

harshitgoel96, 2019-01-24:
Hi 
I have created a script that uses the qmk api to index all the keyboards that are supported by the firmware.
It generates a index.md file which has the keyboard name, link the the keyboard folder, and the description of the keyboard.
It will make it possible to have a dynamic index and update it once a keyboard is removed, added or moved.

The script is available at the following repo
https://github.com/harshitgoel96/QMK-Keyboard-Indexer

tagging @yanfali  @skullydazed

drashna, 2019-01-24:
Actually, Travis CI does this automatically on each merge/commit to the master repo:
https://qmk.fm/keyboards/

But that's neat, still.

harshitgoel96, 2019-01-24:
This is more for the github repo itself... Its getting hard to find the keyboard in the firmware repo.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc