qmk/qmk_firmware #4950: Unable to flash new Massdrop Preonic v3

ElDifinitivo, 2019-01-26:
I just received the new preonic and it seems I have succesfully compiled my custom keymapping. However, the keyboard should be a rev3, but trying to compile anything other than rev2 fails with this error:

```
$~ sudo make preonic/rev3:custom

tmk_core/chibios.mk:89: lib/chibios/os/hal/lib/streams/streams.mk: No such file or directory
make[1]: *** No rule to make target 'lib/chibios/os/hal/lib/streams/streams.mk'.  Stop.
Make finished with errors
make: *** [Makefile:544: preonic/rev3:kristen] Error 1
```
Simply running `sudo make preonic:custom` succesfully compiles a rev2 hex file.

Using the rev2 compiled version though, I am unable to flash the keymapping to the keyboard after clicking the reset button:

```
$~ sudo make preonic:custom:dfu

dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s
```
The keyboard works with the pre-installed default keymapping otherwise, and the keyboard shows up under `lsusb` after clicking reset to flash as:

```
Bus 001 Device 052: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
```

**System Information**
 - Keyboard:  Preonic v3
   - Revision (if applicable):  Recently shipped from Massdrop
 - Operating System: Linux
 - avr-gcc version: 5.4.0

 - arm gcc version: (15:7-2018-q2-4) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 - QMK Firmware version: 0.6.255

drashna, 2019-01-26:
Two issues here:

ChibiOS is probably out of date.  Run `make git-submodule` as that should update the chibiOS stuff.

DFU is not correct flasher.  For ARM, you need to use `dfu-util`, so the full command should be:
```bash
make preonic:default:dfu-util
```

Additionally, you shold not bu using sudo to compile.   It's better to add the bootloader to `udev` rules.
https://docs.qmk.fm/#/faq_build?id=linux-udev-rules

But I think you want:
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE:="0666"
```

ElDifinitivo, 2019-01-27:
Updated ChibiOS and tried using `dfu-util` and got this output:

```
QMK Firmware 0.6.258
Making preonic/rev2 with keymap custom and target dfu-util

make[1]: *** No rule to make target 'dfu-util'.  Stop.
Make finished with errors
make: *** [Makefile:544: preonic:custom:dfu-util] Error 1
```

samirmakwana, 2019-01-28:
Err. You typed rev2 instead of rev3.

ElDifinitivo, 2019-01-31:
Got it now, thanks! It defaults to rev2 and I had forgotten to specifically chose v3.

The hex wasn't compiling in my OP, but after pulling in the newest version and recompiling, I was able to use `dfu-util` to flash.

`make preonic/v3:custom`
`make preonic/v3:custom:dfu-util`

closed 2019-01-31 by ElDifinitivo