qmk/qmk_firmware #4951: Ergodox EZ with WS2812B incorrect colors

LukeeeeBennett, 2019-01-26:
I built an ergodox and added a WS2812B strip.

I am having some issues with the LEDs being the wrong color.

I have modified the ergodox_ez default_osx keymap.

`config.h`

```
#undef RGB_DI_PIN
#undef RGBLED_NUM
#define RGB_DI_PIN D5
#define RGBLED_NUM 8
#define RGBLIGHT_ANIMATIONS
```

`Makefile`

```
RGBLIGHT_ENABLE = yes
```

`rgblight_setrgb_orange` results in: https://i.imgur.com/Q5jMVbU.jpg
`rgblight_setrgb_white` results in: https://i.imgur.com/xB7nrYt.jpg

When I put it into an animation using an RGB keycode, every animation looks the same and is just the one colour per pixel fading in and out, rather than the hue changing as well.

drashna, 2019-02-07:
Well, it should be `rules.mk`, not `makefile`, unless you're using a very old version. 

Additionally, it sounds like the mode may not be set correctly.  But if you're using "setrgb", that shouldn't matter. 

It may be an issue with how the ergodox ez firmware sets up the rgb lights, and you may need to undefine `RGBW_BB_TWI` and `RGBW`

drashna, 2019-04-01:
Was this still an issue?

LukeeeeBennett, 2019-04-02:
Thanks for the info @drashna, I'll try to poke around some more tonight.

I was thinking of looking to your [rgb_layer keymap](https://github.com/qmk/qmk_firmware/tree/master/keyboards/ergodox_ez/keymaps/rgb_layer) for more info aswell.

LukeeeeBennett, 2019-04-02:
I copied your rgb_layer keymap and set `RGB_DI_PIN` to `D5`. The colours still incorrect/weird.

drashna, 2019-04-05:
Did you try undefining the other values? 
Also, have you tried wiping the eeprom? (`EEP_RST` keycode)
And did you set the mode?

drashna, 2019-04-05:
Figured out the issue, actually.

The EZ is designed with RGBW lights in mind, and has a define to enable that behavior.  This sends different data than "just" RGB LEDs. 

You need to add `#undef RGBW` in your config.h file to get proper colors sent.

closed 2019-07-29 by LukeeeeBennett