qmk/qmk_firmware #4953: Default Preonic rev 3 file is a hex file. This should be a .bin file.

shmcgrath, 2019-01-26:
The autogen script in QMK Toolbox does not work properly for Preonic v3. It generates a .hex file when it should be a .bin. I believe the [keyboards] page on the docs also points to an incorrect download file.
[image]

samirmakwana, 2019-01-28:
Noticed that on my system too. Windows 10 64bit. But when I used msysw64 in command line to make a file, it generated a .bin file. May be you could give that a shot?

noroadsleft, 2019-01-29:
Need to add Planck rev6 to this issue also.

drashna, 2019-02-08:
Okay, I looked into this.

The issue is that the Travis build scripts create a HEX file for this, no matter what the normal output would be.  Also, the scripts only look for HEX files, not BIN.  

I've created a PR that should address both issues.

The caveat is that both the qmk.fm website, and the toolbox may need to be updated to use these files properly.

PR #5077 (2019-02-13): Fix AUTOGEN and Travis Compiled Push scripts 

## Description
Currently, when you add `AUTOGEN=true` to the make command, it compiles the file as a `hex` regardless of the architecture.  This is wrong, and generates `hex` files for ARM boards like the Planck Rev6.  This file isn't flashable, as far as I'm aware. 

This changes the AUTOGEN script to use the extension agnostic `FIRMWARE_FORMAT` instead of hard coding it to `hex`.   This means that the correct `bin` file is copied to the keymap folder, instead of a `hex`, when appropriate.

This also changes the Travis Compiled script to look for [... cut]
closed 2019-02-13 by drashna