qmk/qmk_firmware #4958: DZ60: Board Initializes to Wrong Layer

bhilburn, 2019-01-27:
When I first plug in / power in my DZ60, it always starts in the third of my four layers. I'm not sure why it does this, and I can't seem to figure out how to actually just make it start in the base layer.

**System Information**
 - Keyboard:  DZ60
   - Revision (if applicable):  USB-C
 - Operating System: Linux
 - avr-gcc version: 6.3.0

 - arm gcc version: 7.1.0

 - QMK Firmware version: 0.6.256

 - Any keyboard related software installed? No
   - [ ] Auto Hot Key
   - [ ] Karabiner
   - [ ] Other

**Additional context**

You can find my code here:
[my keymap.c](https://github.com/bhilburn/qmk_firmware/blob/cstd60-keymap/keyboards/dz60/keymaps/cstd60/keymap.c)

pelrun, 2019-01-27:
Have you tried resetting the eeprom using QMK Toolbox?

bhilburn, 2019-01-27:
I have not tried resetting the eeprom, no. I'll give that a go - thanks for
the suggestion!

drashna, 2019-02-08:
Did that fix the issue?

bhilburn, 2019-02-08:
I actually haven't gotten a chance to try it yet, actually. I'll update this issue as soon as I do, hopefully this weekend.

germ, 2019-05-03:
Just had a customer have this exact problem, resetting the EEPROM from toolbox was the correct answer.

bhilburn, 2019-05-04:
Can confirm it fixed it for me, too! Closing this one out =)

closed 2019-05-04 by bhilburn