qmk/qmk_firmware #4959: Preonic Rev3: LALT and LGUI Keys swapped

Boy-314, 2019-01-27:
**After flashing my keymap for my preonic rev3 with a 2U right spacebar, the left alt and left windows keys have swapped functionality**

**System Information**
 - Keyboard:  Preonic
   - Revision (if applicable):  3
 - Operating System: Windows 10
 - avr-gcc version: 4.9.2

 - arm gcc version: 6.3.1

 - QMK Firmware version: 0.6.180

 - Any keyboard related software installed? 
   - [ X ] Auto Hot Key, but not used
   - [ ] Karabiner
   - [ ] Other

**Additional context**

 
Tried: resetting eeprom with the file provided by drashna on the qmk discord. link to download here: http://puu.sh/CCXF8/bdb582d2e7.bin

Fix: Swap the LALT and LGUI keycodes on my layout.

13-bit, 2019-01-27:
There's two controls on the adjust layer that might help you. 'Lower + Raise + G' will set it to Mac mods (Alt, then GUI). 'Lower + Raise + H' will set it to Win mods.

drashna, 2019-01-27:
^

Resetting the EEPROM *should* fix this, though.

closed 2019-02-08 by drashna