qmk/qmk_firmware #4968: Make SSD1036 independent from i2c implementation

thomasbaart, 2019-01-27:
## Feature Request Type

- [ ] Core Functionality
- [x] Add-on hardware support (e.g. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing Feature(s)
- [ ] New behavior

## Description 

I'm writing a guide on how to use and customize OLED display functionality, since it's currently undocumented. I made [the following commit in my fork](https://github.com/thomasbaart/qmk_firmware/commit/7d2b52f96d80e415060b21ce44439aad3f035a4d) that gets a "Hello world" sample up and running.

What I had to do, in a nutshell:

* I had to set `SPLIT_KEYBOARD = yes` in `rules.mk`
* I had to add `SRC += ssd1036.c` to `rules.mk`
* I had to add defines for `SSD1306OLED` and `USE_I2C` to my `config.h`
* I had to implement `matrix_master_OLED_init(void)` in my `keymap.h`, including `#include "split_util.h"`
* For the rest it's expected: `implement matrix_scan_user(void)` and call `iota_graphics_task`, then implement `iota_graphics_task_user(void)` and add some calls like `iota_gfx_clear_screen()` followed by `iota_gfx_write("Hello world!")`.

My breadboard setup on which I'm testing the sample is a single Pro Micro, and thus is not a split board. I added a dependency on the split code since that seems to be the only place where an official QMK implementation exists of `i2c.c` and `i2c.h`.

This dependency seems undesirable. Without much background knowledge, it might be a possible solution to include an official version of these files that are standalone when the keyboard is not split yet depends on i2c code.

@pelrun mentioned having to use `i2c_master.c`. While that could be a solution, `ssd1036.c` explicitly calls for `i2c.h`, which `i2c_master.h` is not.

ishtob, 2019-01-28:
The current implementation was built around having ssd1306 as a feature in a keyboard folder, I think ideally we should move the task into quantum.c and put a ifdef around it.

ishtob, 2019-01-29:
#4994 has been opened, the current commit in there should make the SSD1306 use i2c_master.c, if you have a board you can test it on, it would be most welcomed

thomasbaart, 2019-01-29:
I made a new commit removing the dependencies and built the project by executing `make thomasbaart_oled`. It won't build, and I think it does not have to do with my code, looking at the logs.

Maybe I'm missing potential `SRC +=` statements in my `rules.mk`?

If I remove `SRC += ssd1306.c`, the compiler doesn't know about the iota functions. I tried adding `SRC += i2c_master.c`, but that does not help.

Here are the logs from `make`:

```
$ make thomasbaart_oled
QMK Firmware 0.6.258
Making thomasbaart_oled with keymap default

avr-gcc.exe (AVR_8_bit_GNU_Toolchain_3.6.1_1752) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   23966       0   23966    5d9e .build/thomasbaart_oled_default.hex

Compiling: drivers/avr/ssd1306.c                                                                   drivers/avr/ssd1306.c: In function '_send_cmd1':
drivers/avr/ssd1306.c:37:7: error: too few arguments to function 'i2c_start'
   if (i2c_start(SSD1306_ADDRESS)) {
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:21:14: note: declared here
 i2c_status_t i2c_start(uint8_t address, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:42:7: error: too few arguments to function 'i2c_write'
   if (i2c_write(0x0 /* command byte follows */)) {
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:48:7: error: too few arguments to function 'i2c_write'
   if (i2c_write(cmd)) {
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:54:3: error: too few arguments to function 'i2c_stop'
   i2c_stop();
   ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:29:14: note: declared here
 i2c_status_t i2c_stop(uint16_t timeout);
              ^
drivers/avr/ssd1306.c: In function 'clear_display':
drivers/avr/ssd1306.c:91:7: error: too few arguments to function 'i2c_write'
   if (i2c_write(SSD1306_ADDRESS)) {
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:94:7: error: too few arguments to function 'i2c_write'
   if (i2c_write(0x40)) {
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:100:7: error: too few arguments to function 'i2c_write'
       i2c_write(0);
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:107:3: error: too few arguments to function 'i2c_stop'
   i2c_stop();
   ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:29:14: note: declared here
 i2c_status_t i2c_stop(uint16_t timeout);
              ^
drivers/avr/ssd1306.c: In function 'matrix_render':
drivers/avr/ssd1306.c:275:7: error: too few arguments to function 'i2c_write'
   if (i2c_write(SSD1306_ADDRESS)) {
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:278:7: error: too few arguments to function 'i2c_write'
   if (i2c_write(0x40)) {
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:289:9: error: too few arguments to function 'i2c_write'
         i2c_write(colBits);
         ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:293:7: error: too few arguments to function 'i2c_write'
       i2c_write(0);
       ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:22:14: note: declared here
 i2c_status_t i2c_write(uint8_t data, uint16_t timeout);
              ^
drivers/avr/ssd1306.c:300:3: error: too few arguments to function 'i2c_stop'
   i2c_stop();
   ^
In file included from drivers/avr/ssd1306.c:4:0:
drivers/avr/i2c_master.h:29:14: note: declared here
 i2c_status_t i2c_stop(uint16_t timeout);
              ^
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:379: .build/obj_thomasbaart_oled_default/ssd1306.o] Error 1
Make finished with errors
make: *** [Makefile:544: thomasbaart_oled] Error 1
```

ishtob, 2019-01-29:
sorry it was my fault, I failed to realize the i2c codes needs some more changes/arguments to work. @pelrun helped with s;me input on how to proceed and I just pushed and replaced the commits from yesterday. I am able to `make hadron/ver2:default` without any errors

you may have to replace the `i2c_master_init()` with `i2c_init()` in your keymap

XScorpion2, 2019-02-24:
Thomas, take a look at this pr: https://github.com/qmk/qmk_firmware/pull/5151
Most everything you mentioned should be self contained within the ssd1306.c source and only uses OLED_ENABLED

thomasbaart, 2019-02-24:
While that does work, what issue intends to accomplish is to rely on the native SSD1306 support which QMK has built in, instead of relying on a separate file for each keyboard. That should make it easier to add new functionaly across all versions, and should streamline I2C code as well.

XScorpion2, 2019-02-24:
That PR does just that, it centralizes all the keymaps to use the ssd1306 in the "drivers/avr/ssd1306.c" folder

thomasbaart, 2019-02-24:
My apologies. I was looking on mobile and skimmed through it too quickly. Nice work, thank you very much!

Maybe we should make a documentation page for the SSD1306 support, so people will know to add that snippet into their makefile.

XScorpion2, 2019-02-24:
Ya, documentation would be nice, same with more folks testing this! =D

XScorpion2, 2019-02-28:
Added docs, please review when you have some time: https://github.com/qmk/qmk_firmware/pull/5151/files?short_path=4cd9fbb#diff-4cd9fbbdaffa495d01004e2a1983dbc5

drashna, 2019-11-21:
This is fixed now. Closing.

closed 2019-11-21 by drashna