qmk/qmk_firmware #4991: Mod tap sends mod keycode when key is tapped instead of the default keycode

lalicari, 2019-01-29:
Just built my Preonic rev3 that I got in the mail yesterday. I'm trying to make the enter key right shift when held and enter when tapped. I've tried RSFT_T(KC_ENT) and MT(MOD_RSFT, KC_ENT) in my QWERTY keymap, but neither seem to work. 

Tapping enter inputs right shift and holding also inputs right shift. Expected result is to have right shift when held and enter when tapped.

 - Keyboard:  Preonic
   - Revision (if applicable):  Rev 3
 - Operating System: macOS Mojave 10.14.2
 - avr-gcc version: avr-gcc (GCC) 7.3.0
 - arm gcc version: 7.3.1 20180622
 - QMK Firmware version: 0.6.121
 - Any keyboard related software installed? 
   - [ ] Auto Hot Key
   - [X] Karabiner
   - [ ] Other

**Additional context**
QMKToolbox crashes when I try loading the firmware this way, so I load it on the command line via:

`sleep 10 && make preonic/rev3:lalicari:dfu-util`

I had to `make git-submodule` and `git submodule update --init lib/chibios` before I was able to get anything working.

lalicari, 2019-02-04:
Can anyone get a repro on this? Just want to make sure I'm not crazy 😅

kandread, 2019-02-04:
Have the same issue as you (same OS etc), ended up putting enter on the right space key (using the right thumb for it) for now at least.

lalicari, 2019-02-05:
I might have time this weekend to take a look at this issue. Do any experienced contributors know if this might be a good issue for a newb like me to get started with?

drashna, 2019-02-08:
Sounds like you might need `Tapping Force Hold`
https://docs.qmk.fm/#/feature_advanced_keycodes?id=tapping-force-hold

roguepullrequest, 2019-03-23:
Having a similar problem with planck/rev4 with `MT(MOD_LCTL | MOD_LSFT, KC_LGUI)` or `C_S_T(KC_LGUI)`. 

Only `MOD_LCTL | MOD_LSFT` are sent. Regardless if its tapped or held.

lalicari, 2019-08-05:
Rebased my repo and things seem to be working now with KC_SFTENT

closed 2019-08-05 by lalicari