qmk/qmk_firmware #4997: Rotary encoder on Preonic rev3 does not work

adiron, 2019-01-29:
As of commit 8da9d3330e7904c57b240cfb1b2d3312ad81ccf9 (as per a `git bisect` I ran) the rotary encoder does not work with the default Preonic keymap (or my own for that matter).

**System Information**
 - Keyboard:  Preonic
   - Revision (if applicable):  Rev3
 - Operating System:
macOS Mojave
 - avr-gcc version: 
avr-gcc (GCC) 7.3.0
 - arm gcc version: 
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

adiron, 2019-01-30:
Also fwiw, in 8da933's parent, I can't get dual role keys to work

drashna, 2019-02-02:
I've let jack know about this, since ... well, it's his keyboard.

If you're compiling from source, you can revert that commit using git, for the time being.

adiron, 2019-02-03:
Thank you very much :)

closed 2019-02-07 by adiron