qmk/qmk_firmware #5022: Tap Dance compile error

pvinis, 2019-01-31:
I am trying to compile with Tap Dance, and I get this:
```
quantum/process_keycode/process_tap_dance.h:37:29: error: integer overflow in expression [-Werror=overflow]
 #define TD(n) (QK_TAP_DANCE + n)
                             ^
```

**Describe the bug**

**System Information**
 - Keyboard:  
   - Revision (if applicable):  
 - Operating System:
 - avr-gcc version: 

 - arm gcc version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] Auto Hot Key
   - [ ] Karabiner
   - [ ] Other

**Additional context**

 
Probably @algernon will have some idea.

algernon, 2019-01-31:
How many tap-dance keys do you have? Can you share the values you pass to `TD()`? (the enum)

My first suspicion is that you're using something like:

```c
enum {
 SOMETHING = SAFE_RANGE,
 FOO,
 BAR
};

// TD(FOO)
```

You don't need the `SAFE_RANGE` stuff. If it's not that, if you can share your keymap, that'd be helpful.

drashna, 2019-02-08:
Were you still having issues with this? 

If so, could you post a link to the code  that you're using (gist, or github)

pvinis, 2019-02-08:
I'll check it today. Sorry for being late.

pvinis, 2019-02-08:
Hm, @algernon indeed I had the `SAFE_RANGE`.
```

// my own keycodes
enum userspace_custom_keycodes {
  PV_ = SAFE_RANGE,

  PV_VRSN, // prints firmware version
  PV_MAKE, // prints the make command of the keyboard
  PV_FLSH, // resets keyboard
  PV_KTMR, // play katamari music

  // tap dance
  TD_FLSH, // flash keyboard (as if the physical flash key was pressed)

  PV_SAFE_RANGE, // used for extra keycodes in the individual keymaps
};

#define TD_3FLS  TD(TD_FLSH)        // tap dance 3 times for flash
```

Without the `SAFE_RANGE` line, it compiles fine.

I thought we should use the safe range to avoid messing up other keycodes?

algernon, 2019-02-08:
Yes, but tap-dance indexes aren't custom keycodes. The issue in your case is that you're mixing custom keycodes with tap-dance indexes, they should be separate enums. Tap-dance "names" are like the macro names, they're not custom keycodes, the `TD()` macro turns the name/index into a keycode, but it's a QMK keycode in the end.

IOW, this is what you want:

```c
// my own keycodes
enum userspace_custom_keycodes {
  PV_ = SAFE_RANGE,

  PV_VRSN, // prints firmware version
  PV_MAKE, // prints the make command of the keyboard
  PV_FLSH, // resets keyboard
  PV_KTMR, // play katamari music

  PV_SAFE_RANGE, // used for extra keycodes in the individual keymaps
};

enum tap_dance_indexes {
  // tap dance
  TD_FLSH, // flash keyboard (as if the physical flash key was pressed)
};

#define TD_3FLS  TD(TD_FLSH)        // tap dance 3 times for flash
```

pvinis, 2019-02-08:
Ah, makes sense. Thanks for explaining. I'll change to that. 👍

closed 2019-02-08 by pvinis