qmk/qmk_firmware #5024: Build Errors

IllusiveBagel, 2019-01-31:
**Describe the bug**
Errors While Trying to Compile a Custom Keyboard and Keymap

**System Information**
 - Keyboard:  Custom  
 - Operating System: Windows 10 (Using Ubuntu SubSystem)
 - avr-gcc version: 5.4.0

 - QMK Firmware version: 0.6.265

**Additional context**

 
[Gist of keymap.c](https://gist.github.com/IllusiveBagel/a88c88c1b14d1d69e9bc31f59eeffe66)

Error Recieved
```
QMK Firmware 0.6.265
Making Norbit with keymap default

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/Norbit/keymaps/default/keymap.c                                               <command-line>:0:16: error: empty filename in #include
keyboards/Norbit/keymaps/default/keymap.c:3:24: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘keymaps’
 const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
                        ^
keyboards/Norbit/keymaps/default/keymap.c:20:24: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘fn_actions’
 const uint16_t PROGMEM fn_actions[] = {
                        ^
 [ERRORS]
 |
 |
 |
tmk_core/rules.mk:376: recipe for target '.build/obj_Norbit_default/keyboards/Norbit/keymaps/default/keymap.o' failed
make[1]: *** [.build/obj_Norbit_default/keyboards/Norbit/keymaps/default/keymap.o] Error 1
Makefile:529: recipe for target 'Norbit:default' failed
Make finished with errors
make: *** [Norbit:default] Error 1
```

fauxpark, 2019-01-31:
I think you may be missing some files. You should also have a `Norbit.h`, `config.h` and `rules.mk` in your keyboard folder. I recommend using `util/new_project.sh` to generate this automatically, see https://docs.qmk.fm/#/hardware_avr?id=adding-your-avr-keyboard-to-qmk

You should also change the `F(0)` in your first layer to `TG(1)`, and the one on the second layer to `KC_TRNS` or `_______`. `fn_actions` is deprecated.

IllusiveBagel, 2019-02-01:
> I think you may be missing some files. You should also have a `Norbit.h`, `config.h` and `rules.mk` in your keyboard folder. I recommend using `util/new_project.sh` to generate this automatically, [...]

I have those files too but the error is being generated in the keymap.c so didn't think they needed to be included and thanks for the info on the 'TG(1)' stuff i'll change that

fauxpark, 2019-02-01:
There doesn't seem to be anything wrong with the keymap. I tried generating a new keyboard called Norbit and pasting it in, and it's building fine for me on MSYS.

The specific error here is `empty filename in #include`, which means that `QMK_KEYBOARD_H` is `""` for some reason. I don't 100% understand exactly how it is derived but it should be evaluating to your `Norbit.h`.

IllusiveBagel, 2019-02-01:
Here is the Norbit.h file if that helps

[Gist Link](https://gist.github.com/IllusiveBagel/19050ced8dc7cfe87b6205c34dfe1266)

fauxpark, 2019-02-01:
There's a missing `\` after the `LAYOUT(` define. Also, keys that are defined in your `LAYOUT` matrix to be `KC_NO` don't need to be included in the keymap. Lastly, the include guards can be replaced with `#pragma once`.

Here's what these files should look like:

```c
#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
  LAYOUT(
    TG(1),   KC_PSLS, KC_PAST, KC_PMNS,
    KC_KP_7, KC_KP_8, KC_KP_9,
    KC_KP_4, KC_KP_5, KC_KP_6, KC_PPLS,
    KC_KP_1, KC_KP_2, KC_KP_1, KC_PENT,
             KC_KP_0, KC_PDOT
  ),
  LAYOUT(
    _______, KC_PSLS, KC_PAST, KC_PMNS,
    KC_KP_7, KC_KP_8, KC_KP_9,
    KC_KP_4, KC_KP_5, KC_KP_6, KC_PPLS,
    KC_KP_1, KC_KP_2, KC_KP_1, KC_PENT,
             KC_KP_0, KC_PDOT
  )
};
```

```c
#pragma once

#include "quantum.h"

#define LAYOUT( \
  K00, K01, K02, K03, \
  K10, K11, K12,      \
  K20, K21, K22, K23, \
  K30, K31, K32, K33, \
       K41, K42       \
) { \
  { K00,   K01,   K02,   K03   }, \
  { K10,   K11,   K12,   KC_NO }, \
  { K20,   K21,   K22,   K23   }, \
  { K30,   K31,   K32,   K33   }, \
  { KC_NO, K41,   K42,   KC_NO }  \
}
```

And a very minimal `config.h` and `rules.mk`:

```c
#pragma once

#include "config_common.h"

#define VENDOR_ID       0xFEED
#define PRODUCT_ID      0x0000
#define DEVICE_VER      0x0001
#define MANUFACTURER    You
#define PRODUCT         Norbit
#define DESCRIPTION     A custom keyboard

#define MATRIX_ROWS 5
#define MATRIX_COLS 4

#define MATRIX_ROW_PINS { D0, D1, D2, D3, D4 }
#define MATRIX_COL_PINS { B0, B1, B2, B3 }
#define UNUSED_PINS

#define DIODE_DIRECTION COL2ROW
```

```make
MCU = atmega32u4
F_CPU = 16000000
ARCH = AVR8
F_USB = $(F_CPU)
OPT_DEFS += -DINTERRUPT_CONTROL_ENDPOINT
BOOTLOADER = atmel-dfu
```

I still was not able to reproduce the same error, though.

IllusiveBagel, 2019-02-01:
Ok I deleted it and started again and got it compile this time. Thanks for the help

closed 2019-02-01 by IllusiveBagel