qmk/qmk_firmware #5037: Split i2c animations cause random key losses and insertions

Lenbok, 2019-02-02:
Some time between 0.6.233 and 0.6.263 RGB underglow animations on the slave side of split i2c keyboards were enabled (previously only the master side would animate).

When an animated mode is turned on, I am experiencing the slave side of the keyboard missing keypresses and even injecting keypresses while no other typing is going on. Turning off the animation makes the keyboard behave normally. Unplugging the slave side makes the master behave normally.

(Additionally, if I switch to an animated mode and toggle the RGB to off, then unplug and re-plug the keyboard, I get the lost/injected keypresses even through the LEDs are not active - this suggests that on keyboard startup the animation is running even though the LEDs are off)

@mtei @pelrun 

**System Information**
 - Keyboard:  Redox
   - Revision (if applicable):  handwire
 - Operating System: Ubuntu 18.04
 - avr-gcc version: 5.4.0
 - arm gcc version: 6.3.1
 - QMK Firmware version: 0.6.263

drashna, 2019-02-02:
Make sure you've flashed both sides

Lenbok, 2019-02-02:
Yes, I always flash both sides.

Lenbok, 2019-02-03:
@nooges noted in his previous experiments with i2c split animations that the key for the upper left key on slave would fire off on its own a bunch with the split RGB stuff. 

Since that key is backspace on my keyboard, that accounts for what I'm seeing (my "lost keys" are just those getting immediately backspace removed). I guess when the animations are running something is splatting part of the matrix or more likely corrupting the first element of matrix data being transmitted/received?

pelrun, 2019-02-03:
Sounds like they're colliding in i2c address space, although I can't see how that's happening (RGB starts at 1 and is only 4 bytes long, the matrix starts at 6.) Could be a bug in the i2c implementation...

Does anything change if you modify I2C_KEYMAP_START (in split_common/i2c.h currently) to 8?

Lenbok, 2019-02-03:
@pelrun Bumping I2C_KEYMAP_START to 8 and flashing both sides did not help.

pelrun, 2019-02-03:
The answer is it *doesn't* fit, and there's even a comment to that effect. The speed value gets lost off the end. Fine for non-split kbs (they just don't retain the speed across reboots) but split kb's will have independent values on each side.

Lenbok, 2019-02-03:
OK, I see, the comment about EEPROM confused me since we were looking into i2c transport stuff. (Looks like RGB underglow doesn't ever use the speed field anyway, it only seems to be used by RGB matrix code.)

Lenbok, 2019-02-03:
If I go into the animation code (e.g. rgblight_effect_snake) and comment out just the final call to rgblight_set() so the rest of the function is running, then I don't get any spurious keystrokes from that animation mode, so it seems the interference is coming from the ws2812 sending code. Changing RGB_NUM on the firmware flashed to the slave side seems to change which keys are getting randomly pressed. Maybe some incompatibility with how the i2c and ws2812 are dealing with interrupts?

Lenbok, 2019-02-03:
@pelrun Since you're reworking the split i2c stuff in #4974 it's probably not worth worrying about it too much here (unless the same problem happens with your PR :-) )

nooges, 2019-02-05:
Possibly related issue: https://reddit.com/r/olkb/comments/an46xu/phantom_key_presses_from_quefrency_left_side_over/

Lenbok, 2019-02-07:
And here: https://www.reddit.com/r/MechanicalKeyboards/comments/ao1z6u/qmk_rgb_animations_are_summoning_dark_forces_or/

unrau, 2019-02-07:
I am having a similar issue, with some extra icing on top.

@Lenbok mentioned that @nooges reported that the upper-left key of their slave PCB was repeatedly being returned while playing an RGB animation. That is consistent with my experience, with the addition of `'` or `^[` being returned for me as well.

Details of my troubleshooting:

If I leave an animation playing with a text editor in focus, the typing bar remains solid instead of flashing, as though `ESC` were constantly being returned (which it is). Also, sporadically the `'` character is returned into the text editor. In my DVORAK layout, the `'` key is right beside the `ESC` key, and `ESC` is the top-left key on the left PCB, which is my slave PCB.

If I leave an animation playing with a terminal in focus, running showkey, it shows keycode 1 (`ESC`) being pressed and released repeatedly. No `'` characters are recorded by showkey (which is keycode 40), but interestingly, the characters `^[` show up at the same frequency as `'` showed up in the text editor. This makes me think that `CTRL`+`[` is being returned... but showkey never actually records those keystrokes, only the `ESC` keystrokes. I attached a screenshot of this.

bleything, 2019-02-12:
@unrau: the `^[` thing is sort of a red herring. It's the same thing as escape, you're just seeing the terminal echoing the escape presses.

bleything, 2019-02-12:
Hello! I'm the person who made the reddit post that @nooges [mentioned last week](https://github.com/qmk/qmk_firmware/issues/5037#issuecomment-460629458).

I just tried reverting to 0.6.233 to see if my problem would go away. Instead, I'm having an exciting new problem: when in i2c mode, the left side of the keyboard no longer works over TRRS. Keypresses are not registered. In serial mode it works fine. Hilariously, RGB animations work fine in both modes. If I plug the left side in over USB it works correctly.

So I updated to master and tried again. Flashed both sides of the keyboard. Same thing.

Best guess I have is that I jiggled something around enough to create a short somewhere. I won't have time to go over the wiring until next week but I thought I'd mention this here in case it sounded familiar to anyone.

Lenbok, 2019-03-13:
Now #4974 is merged to master this is fixed for me. (At least on my redox - it is probably still an issue for split boards that haven't been migrated to use split_common).

closed 2019-04-05 by drashna

bcat, 2019-04-29:
FWIW, I can still see this on my Quefrency, which should be using split_common, as far as I can tell. It seems intermittent, but serial mode works fine (which makes me think it's not just a short or other physical problem). Should I comment more here, or open a new issue? Thanks!